espressif/esp-aws-iot #122: Memory leak in MDNS found when reconnecting MQTT.

AlexRadek, 2022-04-28:
Hello everyone,

I found a leak when using the ota_mqtt example found in "esp-aws-iot\examples\ota\ota_mqtt". Every time the MQTT connection is lost and reestabilished about 600 bytes are leaked which will lead long term to a stack overflow in the Mdns task.

`I (2713) DEMO: OTA over MQTT demo, Application version 0.9.0
I (2713) DEMO: Disconnected successfully
I (2723) DEMO: Heap Size: 137312
I (2863) wifi_init: WiFi IRAM OP enabled
I (2873) wifi_init: WiFi RX IRAM OP enabled
I (2873) example_connect: Connecting to xxxx
I (61733) DEMO:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (61733) DEMO: OTA over MQTT demo, Application version 0.9.0
I (64713) DEMO: Disconnected successfully
I (64713) DEMO: Heap Size: 86784
I (64713) wifi:state: run -> init (0)
I (64713) wifi:pm stop, total sleep time: 54066304 us / 61783438 us
I (127713) DEMO: Disconnected successfully
...
I (127713) DEMO: Heap Size: 86068
I (190713) DEMO: Disconnected successfully
...
I (190713) DEMO: Heap Size: 85484
I (253713) DEMO: Disconnected successfully
...
I (253713) DEMO: Heap Size: 84708
`

I reduced the log so that it is more readeable and I had to cut some sensitive information. To replicate this, simply disconnect the MQTT and reestablish it in loop. Using esp_get_free_heap_size() you will find that there is a leak.

[ota_demo_core_mqtt.zip](https://github.com/espressif/esp-aws-iot/files/8582527/ota_demo_core_mqtt.zip)

dhavalgujar, 2022-04-28:
Hi @AlexRadek, thank you for reporting this.
We have just pushed out an [update](https://github.com/espressif/esp-aws-iot/commit/ee4449d94d6a3f27a9906d89ecc75e22cc1b051a#diff-8fa307e1d2a487aac77cae6ae9b58899ebb3a6d0add7f579c8662cc97a41b383R44) to the network_transport layer that fixes this.

Can you please try again after pulling the latest changes from the `master` branch and check if you are still seeing this?

AlexRadek, 2022-04-29:
Thank you for answering @dhavalgujar !

At the moment I am in the release/beta brench on the latest commit. I believe the fix you mentioned is already there.