espressif/esp-aws-iot #129: Integrate the esp-aws-iot libraries within a C++ project

ChromaMaster, 2022-07-18:
Hi, 

I'm trying to integrate the `esp-aws-iot` library in a C++ project without any success. The only library that I'm currently aware of is the `coreHTTP` and the esp port of the transport libraries, present in the `libraries/coreHTTP/port/network_transport` directory but I'm sure this would happen as well with the rest of the libraries.

After taking a look at those libraries, I figured out that the version of the `coreHTTP` that the project is using has no support for C++ (it doesn't add the `extern "C"` clause needed to link the functions with C++).

I've also taken a look at the specific version of the `coreHTTP` the `aws-iot` and the `esp-aws-iot` use:
- `aws-iot`: [0fbca6ddabe898f9191c8af1f0285b000d492caf](https://github.com/FreeRTOS/coreHTTP/tree/0fbca6ddabe898f9191c8af1f0285b000d492caf) from Feb 17 2022, 
- `esp-aws-iot`: [b349f89e67f5feef51438dc09ed89959dedc18d5](https://github.com/FreeRTOS/coreHTTP/tree/b349f89e67f5feef51438dc09ed89959dedc18d5) from Dec 15 2020.

Is there any plan to update the third party libraries you are using as well as to update the rest of the projec to to support integration with c++?

Thanks!

davidallenmann, 2022-08-01:
I had the same issue. 
I was able to run the functions in the MQTT mutual auth demo (https://github.com/espressif/esp-aws-iot/tree/master/examples/mqtt/tls_mutual_auth/main) by wrapping the AWS library includes with extern "C" in my C++ code.

This also had to be done in demo_config.h

```
// from MQTT example app_main
extern "C"
{
#include "nvs_flash.h"
#include "esp_event.h"
#include "esp_netif.h"
#include "protocol_examples_common.h"

    // from MQTT example
    /* MQTT API headers. */

#include "core_mqtt.h"
#include "core_mqtt_state.h"

/* OpenSSL sockets transport implementation. */
#include "network_transport.h"

/*Include backoff algorithm header for retry logic.*/
#include "backoff_algorithm.h"

/* Clock for timer. */
#include "clock.h"
}
```

This also had to be done in demo_config.h (shortened version below)
```

/* Include header that defines log levels. */
extern "C" {
    #include "logging_levels.h"
}

/* Logging configuration for the Demo. */
#ifndef LIBRARY_LOG_NAME
    #define LIBRARY_LOG_NAME     "MQTT_DEMO"
#endif
#ifndef LIBRARY_LOG_LEVEL
    #define LIBRARY_LOG_LEVEL    LOG_INFO
#endif

extern "C" {
    #include "logging_stack.h"
}
/************ End of logging configuration ****************/

/**
 * @brief The name of the MQTT library used and its version, following an "@"
 * symbol.
 */
extern "C" {
    #include "core_mqtt.h"
}
#define MQTT_LIB    "core-mqtt@" MQTT_LIBRARY_VERSION

#endif /* ifndef DEMO_CONFIG_H_ */
```

SolidStateLEDLighting, 2022-10-15:
Guys, thanks for the post/response.   This is important information that helped.   Yes, without the extern C declaration, you can get the project to compile but it won't link.   They should a least mention this issue in the documentation in the undefined references section of the Build System documentation page.