espressif/esp-aws-iot #131: Is there any plan for this frame work to support ESP-IDF v4.4 or above?

Kuang-ShengCheng, 2022-08-02:
We are planning to use the ESP32-S3 module, which requires the use of ESP-IDF v4.4 or v5.0, and since it is mentioned that this framework only supports ESP-IDF v4.3 currently, we are not sure if we can use this framework to develop the AWS IoT application on the S3 module.

Rgarrido-RBN, 2022-08-05:
There is an example of use [here](https://github.com/FreeRTOS/iot-reference-esp32c3), you have to do minor changes in ESP-IDF configuration in order to make it work with ESP-IDF v4.4, but I understand that is officially suported.

You have to enable ** mbedTLS threading abstraction layer** from the menuconfig if you are using it in your own project. Unlike the previous version, you could not add it as a usual component on generic component directory of ESP-IDF, you need to add it like the example, as a component of the project. Other difference I found is with this new version you have different components on esp-aws-iot component, so you need to add each of the recursive components in you REQUIRES.

This is the workaround founded, but maybe exists a better solution in order to make it work without setting all this configuration that could have conflicts with some other component of the ESP-IDF.

I'm also waiting if there is some way of integrate this framework in ESP-IDF.

Kuang-ShengCheng, 2022-09-07:
There is an update on the https://github.com/espressif/esp-aws-iot page, saying

```
ESP-IDF can be downloaded from https://github.com/espressif/esp-idf/
Please set your branch to release/v4.3 or release/v4.4 and pull in the latest changes.
IDF release/v5.0 is not supported.
```
Does this mean that ESP-IDF v4.4 is now officially supported without tweaking the latest esp-aws-iot code?