espressif/esp-aws-iot #136: AWS IoT sometimes unable to reconnect

tobyloki, 2022-09-13:
I've bene using the thing shadow example based on https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/release-v3.1.x. Sometimes the device disconnects from AWS and is unable to reconnect even though there is wifi and internet available. It gives out the error NETWORK_ATTEMPTING_RECONNECT. What can I do to make sure the device always reconnects back to AWS?

lawrenceko-intecular, 2022-10-06:
I also have the same issue even with the ota_mqtt demo project, any way to resolve this?

SolidStateLEDLighting, 2022-10-16:
If you disconnect, then I suggest you do a full shutdown of the TLS/MQTT connection, then restart if Wifi/SNTP is still connected and valid.

Be sure that you're always going through  MQTT_ProcessLoop when you are idle and doing nothing because  MQTT_ProcessLoop initiates/handles your PING/PINGRESP.     If you're not continuously moving through MQTT_ProcessLoop, then your connection will get dropped.