espressif/esp-aws-iot #137: OTA Job shows status Failed even after ESP32 successful update

lawrenceko-intecular, 2022-09-20:
Hello,

We have tested the `ota_mqtt` example and everything works fine except the Status showed failed even after ESP32 has successfully updated via OTA.

Are there any messages we need to send back to AWS IoT to confirm the update is successful and complete?

Thank you.

[image]

lawrenceko-intecular, 2022-09-27:
Any updates on this?

davidallenmann, 2022-09-29:
I got the same error until I increased the APP_VERSION_BUILD in demo_config.h

SolidStateLEDLighting, 2022-10-16:
In theory, once you have rebooted, the system should go check for any in-progress jobs.   It will find the uncompleted OTA that you previously were running.   Then, your software would essentially resume that job and send the final update back to AWS which will render that job as successfully complete.     I don't use those examples --- so, I don't know how it is technically done in those example projects.

bhagirathsinh-vaghela, 2023-08-17:
Possible solution: I might be too late and you may have figured it out. I saw the same and observed that it is dependent on `CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE`.

I am not totally sure if my understanding is correct(I don't work on c projects). But this is what I think is happening...
If this is not enabled(not `CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y` in sdkconfig), after the OTA job execution is done and new firmware is loaded to one of ota partition, the otadata on the otapartition needs to get set as `ESP_OTA_IMG_NEW` [but it gets set as `ESP_OTA_IMG_UNDEFINED` instead](https://github.com/espressif/esp-idf/blob/master/components/app_update/esp_ota_ops.c#L119). The very first boot of the new firmware should see this and try to set new otadata as `ESP_OTA_IMG_PENDING_VERIFY` but that doesn't happen and ultimately we never get to setting the [otadata as `ESP_OTA_IMG_VALID`](https://github.com/espressif/esp-idf/blob/master/components/app_update/esp_ota_ops.c#L763) and thus the OTA execution status is not updated as successful.

In my case, I was seeing that second reboot happens which sees the running app version as the same as what AWS OTA job document says and thus marks the job execution as `FAILED` with `"reason":"rejected: 0x00000010"`(log said: `New image is being rejected: Application version of the new image is invalid: OtaErr_t=OtaErrSameFirmwareVersion`).
I hope this helps.