espressif/esp-aws-iot #138: Request for information on support with esp-idf "release/v5.0"

tandelprashant, 2022-09-20:
Hi,

When can we expect the support for this example on ESP-IDF version "release/v5.0".
Is there any patch available or any other way I can get it work for above mentioned version?

Thanks in advance!

dizcza, 2022-12-03:
I have the very same question. It's sad that no one has answered so far.

dhavalgujar, 2022-12-03:
Hi @dizcza, @tandelprashant,
Apologies, I did not mention this issue in the commit message, else you would have been notified automatically.
We have added support for IDF v5.0 in the [`master`](https://github.com/espressif/esp-aws-iot/tree/master) branch.

There will also be a tagged release (`202210.01-LTS`) shortly, which will bump up the AWS IoT libraries to the [latest](https://github.com/FreeRTOS/FreeRTOS-LTS/releases/tag/202210.01-LTS) release alongside support for IDF v5.0 but support for both is available on the `master` branch already.

Hope this helps.

bdpdx, 2022-12-11:
@dhavalgujar I tried to pull the master branch into a new esp-idf v5 project's `components/` directory and when I build it doesn't seem to compile, idf.py reporting:

`[0/1] Re-running CMake...-- Component directory <project_dir>/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added`

The master branch doesn't have a `CMakeLists.txt` file, although some of the release branches do.

How does one configure idf.py to recognize the component from `master` in the absence of `CMakeLists.txt`?

dhavalgujar, 2022-12-11:
Previously, i.e. till [`release/v3.1.x`](https://github.com/espressif/esp-aws-iot/tree/release/v3.1.x), the entire `esp-aws-iot` SDK needed to be included as a component.
However, from [`release/202012.04-LTS`](https://github.com/espressif/esp-aws-iot/tree/release/202012.04-LTS) onwards, each of the libraries is a separate component that you can pull in, based on your needs, by adding the path to the component, in your application's `CMakelists.txt`.

All of the examples in the repo use the manner I just described, to include these individual components.
For example, if you look at [`examples/ota/ota_mqtt/CMakeLists.txt`](https://github.com/espressif/esp-aws-iot/blob/release/202012.04-LTS/examples/ota/ota_mqtt/CMakeLists.txt) in the `release/202012.04-LTS` or `master` branches, you will see that it includes `ota-for-aws-iot-embedded-sdk`, `coreMQTT`, `corePKCS11` and `backoffAlgorithm` as individual components.

Does this help with the issue that you are facing?

bdpdx, 2022-12-11:
@dhavalgujar Yes that answers my question, thank you. I've been using this project for some years on the 3.x branch and haven't had a need to update yet. Hopefully this migration to the latest code goes ok.

JBHunterI, 2022-12-16:
@dhavalgujar, will that tagged release 202210.01-LTS you mentioned above (on December 3rd) with support for IDF v5.0 be available soon?