espressif/esp-aws-iot #159: Can Fleet-Provisioning-for-AWS-IoT-embedded-sdk be added for better FreeRTOS 202210.00-LTS Support?

loganbenda, 2022-12-15:
Hi, 

I'm trying to use this repo in an AWS IoT project which requires AWS Fleet Provisioning (https://github.com/aws/Fleet-Provisioning-for-AWS-IoT-embedded-sdk), which is supported in https://github.com/FreeRTOS/FreeRTOS-LTS/tree/202210.00-LTS.

Is this planned at all?

Thanks!

SolidStateLEDLighting, 2022-12-29:
I will give you an opinion.   Based on what I have seen over the past 2 years, the esp-aws-iot is a breaking away from the aws-iot-device-sdk-embedded-c.     I suspect you will find that the Espressif is going to diverge from AWS and FreeRTOS on a number of issues.   One example is that Espressif will embrace ESP-TSL whereas AWS will encourage OpenSSL for TSL connections.   The ESP-TLS tool will provide a number of features are profitable to Espressif.    

Now, specifically to Fleet Provisioning...   I have been all through that AWS service.   It all boils down to a very simple idea that anyone can follow if you get your hands dirty in the code.   You are just sending and receiving MQTT messages.   You subscribe, publish, handle responses, and then unsubscribe, that is all it is.   Perhaps that's all everything is??? (including Shadow, OTA, etc)

I would have preferred that Espressif publish one big demo of all AWS services together in one translation unit and allowed users to cut out the unnecessary portions.   However, they give you all the services cut up into little pieces thereby suggesting they should be used individually.   I think this is a mistake philosophically, but Espressif wanted to break it down into small digestible pieces I guess.

loganbenda, 2022-12-30:
Thank you for taking the time to write a reply. As I've been digging into the code I've continually run into that same TLS theme when porting over from the fleet provisioning aws-iot-device-sdk-embedded-c example into Espressif. 

I wished they would have published an all inclusive demo too. I guess my best option is to keep plugging away at the individual integration. At least some of the examples provide some direction.

Thanks again!