espressif/esp-aws-iot #161: Unable to compile or open the menuconfig for the tls_mutual_auth example

arslan437, 2023-01-09:
I am using the vs code and the idf extension. idf is installed through the extension and version is 5.0 

I am unable to compile code and I am also not able to open the menuconfig 

here is output what i got when trying to open the menu config. 

```
[SDK Configuration Editor]
Executing action: confserver
Running cmake in directory d:\upwork\avichalid\gate-door-opener\components\esp-aws-iot\examples\mqtt\tls_mutual_auth\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 d:\upwork\avichalid\gate-door-opener\components\esp-aws-iot\examples\mqtt\tls_mutual_auth"...
-- Found Git: C:/esp-idf/tools/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 

[SDK Configuration Editor]
-- ccache will be used for faster recompilation

[SDK Configuration Editor]
-- The C compiler identification is GNU 11.2.0

[SDK Configuration Editor]
-- The CXX compiler identification is GNU 11.2.0

[SDK Configuration Editor]
-- The ASM compiler identification is GNU

[SDK Configuration Editor]
-- Found assembler: C:/esp-idf/tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe

[SDK Configuration Editor]
-- Detecting C compiler ABI info

[SDK Configuration Editor]
-- Detecting C compiler ABI info - done

[SDK Configuration Editor]
-- Check for working C compiler: C:/esp-idf/tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped

[SDK Configuration Editor]
-- Detecting C compile features

[SDK Configuration Editor]
-- Detecting C compile features - done

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info - done

[SDK Configuration Editor]
-- Check for working CXX compiler: C:/esp-idf/tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped

[SDK Configuration Editor]
-- Detecting CXX compile features

[SDK Configuration Editor]
-- Detecting CXX compile features - done

[SDK Configuration Editor]
-- Building ESP-IDF components for target esp32

[SDK Configuration Editor]
---------------------------ERROR--------------------------

CMake Error at C:/esp-idf/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Configuring incomplete, errors occurred!
See also "D:/Upwork/AviChalid/gate-door-opener/components/esp-aws-iot/examples/mqtt/tls_mutual_auth/build/CMakeFiles/CMakeOutput.log".

[SDK Configuration Editor]
---------------------------ERROR--------------------------

    Syntax Warning in cmake code at column 51

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    C:/esp-idf/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  D:/Upwork/AviChalid/gate-door-opener/components/esp-aws-iot/libraries/backoffAlgorithm/CMakeLists.txt:2
  (include):

    include could not find requested file:

  

      D:/Upwork/AviChalid/gate-door-opener/components/esp-aws-iot/libraries/backoffAlgorithm/backoffAlgorithm/backoffAlgorithmFilePaths.cmake

  Call Stack (most recent call first):

    C:/esp-idf/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    C:/esp-idf/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  C:/esp-idf/esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  C:/esp-idf/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:12 (project)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

cmake failed with exit code 1, output of the command is in the d:\upwork\avichalid\gate-door-opener\components\esp-aws-iot\examples\mqtt\tls_mutual_auth\build\log\idf_py_stderr_output_13764 and d:\upwork\avichalid\gate-door-opener\components\esp-aws-iot\examples\mqtt\tls_mutual_auth\build\log\idf_py_stdout_output_13764

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------
```

SolidStateLEDLighting, 2023-01-09:
Looks like you're having CMake problems with your project setup.

Sadly, the examples that are placed into the middle of the AWS IOT library, don't really show you the proper way of organizing a real deliverable project.   You'll need to study the CMake files and read up on how CMake works so you can learn how to construct a real functioning project.

Here is a snapshot of how I believe a normal project should be organized.   Notice that ESP AWS IOT is placed in the components directory.    From the directory of esp-aws-iot -- everything here is copied from github and pasted into my project with no changes in the library at all.

[image]

In my main (which is also a component by default) I have only system-wide related source code there.   All my other areas of concern are placed in their own component.

I had nagging problems for quite with big projects until I got into components organization.

SolidStateLEDLighting, 2023-01-09:
The EXTRA_COMPONENT_DIRS directive does not apply to items that are already inside your components directory.   You'll need to study the CMake tool to get the understand on this.   I have put all my IOT code in a component.   (This code is all the example code snippets combined together).  

From this component I access the ESP-AWS-IOT library with a CMakeLists.txt file.  (inside your code you include the header files you need like backoffAlgorithm.h, etc)

Here is my CMakeLists.txt file for my IOT component

#
FILE(GLOB_RECURSE SOURCES src/iot/*.cpp)
#
#
set(COMPONENT_ADD_INCLUDEDIRS
    include
)
#
#
# Exposes components to both source and header files.
set(IOT_REQUIRES
    coreMQTT
    nvs_flash
    backoffAlgorithm
    ota-for-aws-iot-embedded-sdk
    gate
    camera
    )
#    
#
# Anything that must be exposed to the sources files, but may remain hidden from the header files.
set(IOT_PRIV_REQUIRES
    main
    cbor
    coreHTTP
    coreJSON
    posix_compat
    device-shadow-for-aws-iot-embedded-sdk
)
#
# We link in all our binary certs within the IOT object, because here is where we use those resources.
#
idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS ${COMPONENT_ADD_INCLUDEDIRS}
                       REQUIRES ${IOT_REQUIRES}
                       PRIV_REQUIRES ${IOT_PRIV_REQUIRES}
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/ca/amazon_root_ca1.pem
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/ca/amazon_root_ca2.pem
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/ca/aws_s3_root_ca_cert.pem
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c1/c1_certificate.pem.crt
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c1/c1_private.pem.key
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c1/c1_public.pem.key
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c2/c2_certificate.pem.crt
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c2/c2_private.pem.key
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c2/c2_public.pem.key
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c3/c3_certificate.pem.crt
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c3/c3_private.pem.key
                       EMBED_TXTFILES ${CMAKE_CURRENT_LIST_DIR}/certs/clients/c3/c3_public.pem.key
                      )

arslan437, 2023-01-09:
OK I found the issue. there are submodule in the repository that but I did not init the sub modules due to which I encountreded the errors. Now I used the following command to clone the esp-aws-iot to 

```
git clone -b "release/202012.04-LTS" --recursive https://github.com/espressif/esp-aws-iot
```

Now I am able to open menu config but there are some error in the code 

```
FAILED: esp-idf/posix_compat/CMakeFiles/__idf_posix_compat.dir/clock_esp.c.obj 
ccache C:\Users\M.Arsalan\esp\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -ID:/pst/gate-door-opener/build/config -ID:/pst/gate-door-opener/components/esp-aws-iot/libraries/common/posix_compat -IC:/Users/M.Arsalan/esp/esp-idf/components/newlib/platform_include -IC:/Users/M.Arsalan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/M.Arsalan/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/M.Arsalan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/M.Arsalan/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_hw_support/include -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -IC:/Users/M.Arsalan/esp/esp-idf/components/heap/include -IC:/Users/M.Arsalan/esp/esp-idf/components/log/include -IC:/Users/M.Arsalan/esp/esp-idf/components/soc/include -IC:/Users/M.Arsalan/esp/esp-idf/components/soc/esp32/. -IC:/Users/M.Arsalan/esp/esp-idf/components/soc/esp32/include -IC:/Users/M.Arsalan/esp/esp-idf/components/hal/esp32/include -IC:/Users/M.Arsalan/esp/esp-idf/components/hal/include -IC:/Users/M.Arsalan/esp/esp-idf/components/hal/platform_port/include -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_rom/include -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_rom/include/esp32 -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_rom/esp32 -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_common/include -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_system/include -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_system/port/soc -IC:/Users/M.Arsalan/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/M.Arsalan/esp/esp-idf/components/xtensa/include -IC:/Users/M.Arsalan/esp/esp-idf/components/xtensa/esp32/include -IC:/Users/M.Arsalan/esp/esp-idf/components/lwip/include -IC:/Users/M.Arsalan/esp/esp-idf/components/lwip/include/apps -IC:/Users/M.Arsalan/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/M.Arsalan/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/M.Arsalan/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/M.Arsalan/esp/esp-idf/components/lwip/port/esp32/include/arch -mlongcalls -Wno-frame-address  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=D:/pst/gate-door-opener=. -fmacro-prefix-map=C:/Users/M.Arsalan/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/posix_compat/CMakeFiles/__idf_posix_compat.dir/clock_esp.c.obj -MF esp-idf\posix_compat\CMakeFiles\__idf_posix_compat.dir\clock_esp.c.obj.d -o esp-idf/posix_compat/CMakeFiles/__idf_posix_compat.dir/clock_esp.c.obj -c D:/pst/gate-door-opener/components/esp-aws-iot/libraries/common/posix_compat/clock_esp.c
D:/pst/gate-door-opener/components/esp-aws-iot/libraries/common/posix_compat/clock_esp.c:17:10: fatal error: esp_timer.h: No such file or directory
   17 | #include "esp_timer.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
[3/21] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D D:\pst\gate-door-opener\build\bootloader\esp-idf\esptool_py && C:\Users\M.Arsalan\esp\python_env\idf5.0_py3.8_env\Scripts\python.exe C:/Users/M.Arsalan/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/pst/gate-door-opener/build/bootloader/bootloader.bin"
Bootloader binary size 0x6700 bytes. 0x900 bytes (8%) free.
[4/19] Building C object esp-idf/coreMQTT/CMakeFiles/__idf_coreMQTT.dir/port/network_transport/network_transport.c.obj
FAILED: esp-idf/coreMQTT/CMakeFiles/__idf_coreMQTT.dir/port/network_transport/network_transport.c.obj
ccache C:\Users\M.Arsalan\esp\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -ID:/pst/gate-door-opener/build/config -ID:/pst/gate-door-opener/components/esp-aws-iot/libraries/coreMQTT/coreMQTT/source/include -ID:/pst/gate-door-opener/components/esp-aws-iot/libraries/coreMQTT/coreMQTT/source/interface -ID:/pst/gate-door-opener/components/esp-aws-iot/libraries/coreMQTT/config -ID:/pst/gate-door-opener/components/esp-aws-iot/libraries/common/logging -ID:/pst/gate-door-opener/components/esp-aws-iot/libraries/coreMQTT/port/network_transport -IC:/Users/M.Arsalan/esp/esp-idf/components/newlib/platform_include -IC:/Users/M.Arsalan/esp/esp-idf/comp [... cut ...]
D:/pst/gate-door-opener/components/esp-aws-iot/libraries/coreMQTT/port/network_transport/network_transport.c:14:25: warning: incompatible implicit declaration of built-in function 'strlen' [-Wbuiltin-declaration-mismatch]
   14 |         .cacert_bytes = strlen( pxNetworkContext->pcServerRootCAPem ) + 1,
      |                         ^~~~~~
D:/pst/gate-door-opener/components/esp-aws-iot/libraries/coreMQTT/port/network_transport/network_transport.c:14:25: note: include '<string.h 
' or provide a declaration of 'strlen'
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```

arslan437, 2023-01-09:
I am new to espidf its bit hard for me understand some thing

So as per your advice here is the update make file 

```
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

# set(EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/examples/common_components/protocol_examples_common"
# 						 "${CMAKE_CURRENT_LIST_DIR}/components/esp-aws-iot/libraries/backoffAlgorithm"
# 						 "${CMAKE_CURRENT_LIST_DIR}/components/esp-aws-iot/libraries/coreMQTT"
# 						 "${CMAKE_CURRENT_LIST_DIR}/components/esp-aws-iot/libraries/common/posix_compat"
# 	)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(gate-door-opener)

target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/certs/root_cert_auth.pem" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/certs/client.crt" TEXT)
target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "main/certs/client.key" TEXT)
``` 

Now I am getting this error 

```
In file included from D:/pst/gate-door-opener/main/mqtt_demo_mutual_auth.c:63:
D:/pst/gate-door-opener/main/includes/demo_config.h:37:10: fatal error: logging_levels.h: No such file or directory
   37 | #include "logging_levels.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

for the make you mean I need to create a new make file in the esp-aws-iot folder with the following content 

```
set(COMPONENT_ADD_INCLUDEDIRS
include
)

set(IOT_REQUIRES
coreMQTT
nvs_flash
backoffAlgorithm
ota-for-aws-iot-embedded-sdk
)

set(IOT_PRIV_REQUIRES
main
cbor
coreHTTP
coreJSON
posix_compat
device-shadow-for-aws-iot-embedded-sdk
)
```

SolidStateLEDLighting, 2023-01-09:
In the ESP, everything is a component.   The IDF is comprised of components, your main is a component and everything in the components directory are a components.

You're going to need to do that I did.   Study.   Spend the hours....  Do the work.

K.

avsheth, 2023-01-09:
Hi @arslan437 
For your previously reported issue, as you rightly found out issue was about missing submodule.

For the compilation issue, it seems that esp_timer directory is missing from your esp-idf or CMakeLists.txt within it is somehow missing or deleted? Can you check and confirm ?

arslan437, 2023-01-09:
Hi @avsheth  for the esp_timer.h error I have updated the posix_compat CMakelist.txt below is the updated time. I have added esp_timer are require 

```
idf_component_register(
    SRCS
        "clock_esp.c"
        "semaphore.c"
    INCLUDE_DIRS
        "."
    REQUIRES esp_timer
)
```

for the strlen error I have followed the compiler suggestion and added <string.h in that file. Now it worked. 

after theses fixes example compiles but my code does not 

Do I need to created a new make file the esp-aws-iot folder ? It would be great if you can guide me in the right direction.

arslan437, 2023-01-10:
@avsheth Thanks I did moved to master branch and all the errors are gone. Code compiled with out any issues. 

For future reference I used the following command to clone the repo

```
git clone -b "master" --recursive https://github.com/espressif/esp-aws-iot
``` 
I will close the issue after little bit more testing. 

Thanks everyone for your help.

closed 2023-01-10 by arslan437