espressif/esp-aws-iot #168: Unable to compile when adding esp-aws-iot as component library

law-ko, 2023-02-06:
Hello,

We are trying to add `esp-aws-iot` components by adding the repo as a submodule. However, we are getting build errors even with the clean project.

.gitmodules:
```
[submodule "libraries/esp-aws-iot"]
	path = libraries/esp-aws-iot
	url = https://github.com/espressif/esp-aws-iot.git
```

CMakeLists.txt:
```

# (Automatically converted from project Makefile by convert_to_cmake.py.)

# The following lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

if(DEFINED ENV{HOMEKIT_PATH})
  set(HOMEKIT_PATH $ENV{HOMEKIT_PATH})
else()
  set(HOMEKIT_PATH ${CMAKE_CURRENT_LIST_DIR}/../../..)
endif(DEFINED ENV{HOMEKIT_PATH})

set(EXTRA_COMPONENT_DIRS  ${HOMEKIT_PATH}/components 
                          ${HOMEKIT_PATH}/components/homekit
                          ${CMAKE_CURRENT_LIST_DIR}/../../common
                          
                          "${CMAKE_CURRENT_LIST_DIR}/../../../libraries/esp-aws-iot/libraries/")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(lightbulb)
```

build error log:
```
Executing action: all (aliases: build)
Running ninja in directory /home/espuser/esp-git/esp-homekit-sdk/examples/im/lightbulb-components/build
Executing "ninja all"...
[1/17] Performing build step for 'bootloader'
[1/1] cd /home/espuser/esp-git/esp-homekit-sdk/examples/im/lightbulb-components/build/bootloader/esp-idf/esptool_py && /home/espuser/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/espuser/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/espuser/esp-git/esp-homekit-sdk/examples/im/lightbulb-components/build/bootloader/bootloader.bin
Bootloader binary size 0x61f0 bytes. 0xe10 bytes (13%) free.
[3/15] Building C object esp-idf/corePKCS11/CMakeFile...S11/source/portable/mbedtls/core_pkcs11_mbedtls.c.obj
FAILED: esp-idf/corePKCS11/CMakeFiles/__idf_corePKCS11.dir/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c.obj 
/home/espuser/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/dependency/3rdparty/pkcs11 -I/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/include -I/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/config -I/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/port -I/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/common/logging -I/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/dependency/3rdparty/mbedtls_utils -I/home/espuser/esp/esp-idf/components/newlib/platform_include -I/home/espuser/esp/esp-idf/components/freertos/include -I/home/espuser/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/espuser/esp/esp-idf/components/freertos/port/xtensa/include -I/home/espuser/esp/esp-idf/components/freertos/include/esp_additions -I/home/espuser/esp/esp-idf/components/esp_hw_support/include -I/home/espuser/esp/esp-idf/components/esp_hw_support/include/soc -I/home/espuser/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/espuser/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/espuser/esp/esp-idf/components/heap/include -I/home/espuser/esp/esp-idf/components/log/include -I/home/espuser/esp/esp-idf/components/lwip/include/apps -I/home/espuser/esp/esp-idf/components/lwip/include/apps/sntp -I/home/espuser/esp/esp-idf/components/lwip/lwip/src/include -I/home/espuser/esp/esp-idf/components/lwip/port/esp32/include -I/home/espuser/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/espuser/esp/esp-idf/components/soc/include -I/home/espuser/esp/esp-idf/components/soc/esp32/. -I/home/espuser/esp/esp-idf/components/soc/esp32/include -I/home/espuser/esp/esp-idf/components/hal/esp32/include -I/home/espuser/esp/esp-idf/components/hal/include -I/home/espuser/esp/esp-idf/components/hal/platform_port/include -I/home/espuser/esp/esp-idf/components/esp_rom/include -I/home/espuser/esp/esp-idf/components/esp_rom/include/esp32 -I/home/espuser/esp/esp-idf/components/esp_rom/esp32 -I/home/espuser/esp/esp-idf/components/esp_common/include -I/home/espuser/esp/esp-idf/components/esp_system/include -I/home/espuser/esp/esp-idf/components/esp_system/port/soc -I/home/espuser/esp/esp-idf/components/esp_system/port/public_compat -I/home/espuser/esp/esp-idf/components/esp32/include -I/home/espuser/esp/esp-idf/components/xtensa/include -I/home/espuser/esp/esp-idf/components/xtensa/esp32/include -I/home/espuser/esp/esp-idf/components/driver/include -I/home/espuser/esp/esp-idf/components/driver/esp32/include -I/home/espuser/esp/esp-idf/components/esp_pm/include -I/home/espuser/esp/esp-idf/components/esp_ringbuf/include -I/home/espuser/esp/esp-idf/components/efuse/include -I/home/espuser/esp/esp-idf/components/efuse/esp32/include -I/home/espuser/esp/esp-idf/components/vfs/include -I/home/espuser/esp/esp-idf/components/esp_wifi/include -I/home/espuser/esp/esp-idf/components/esp_event/include -I/home/espuser/esp/esp-idf/components/esp_netif/include -I/home/espuser/esp/esp-idf/components/esp_eth/include -I/home/espuser/esp/esp-idf/components/tcpip_adapter/include -I/home/espuser/esp/esp-idf/components/esp_phy/include -I/home/espuser/esp/esp-idf/components/esp_phy/esp32/include -I/home/espuser/esp/esp-idf/components/esp_ipc/include -I/home/espuser/esp/esp-idf/components/app_trace/include -I/home/espuser/esp/esp-idf/components/esp_timer/include -I/home/espuser/esp/esp-idf/components/mbedtls/port/include -I/home/espuser/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/espuser/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/espuser/esp/esp-idf/components/nvs_flash/include -I/home/espuser/esp/esp-idf/components/spi_flash/include -I/home/espuser/esp/esp-idf/components/bootloader_support/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/home/espuser/esp-git/esp-homekit-sdk/examples/im/lightbulb-components=. -fmacro-prefix-map=/home/espuser/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4-253-g5ff09f0a06\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/corePKCS11/CMakeFiles/__idf_corePKCS11.dir/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c.obj -MF esp-idf/corePKCS11/CMakeFiles/__idf_corePKCS11.dir/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c.obj.d -o esp-idf/corePKCS11/CMakeFiles/__idf_corePKCS11.dir/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c.obj   -c /home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c
/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:275:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xMutex;                    /**< @brief Mutex that protects write operations to the xObjects array. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:288:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xSessionMutex;     /**< @brief Mutex that protects write operations to the pxSession array. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:306:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xVerifyMutex;      /**< @brief Protects the verification key from being modified while in use. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/espuser/esp-git/esp-homekit-sdk/libraries/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:310:5: error: unknown type name 'mbedtls_threading_mutex_t'
[... cut ...] _free'; did you mean 'mbedtls_mpi_free'? [-Werror=implicit-function-declaration]
         mbedtls_mutex_free( &xP11Context.xObjectList.xMutex );
         ^~~~~~~~~~~~~~~~~~
         mbedtls_mpi_free
cc1: some warnings being treated as errors
[5/15] Building C object esp-idf/ota-for-aws-iot-embe...ir/ota-for-aws-iot-embedded-sdk/source/ota_mqtt.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

None of the project codes are being edited. How to fix this?

Thank you.

SolidStateLEDLighting, 2023-02-16:
By default, the components directory will be automatically registered as a place to find components.   Place the esp-aws-iot library right into your components directory.

In your main directory (another component by default) be sure to correctly apply your REQUIRES variables to include all of the items that you need from esp-aws-iot.   It will look a it like this inside your CMakeLists.txt...

# Exposes components to both source and header files.
set(REQUIRES
    coreMQTT
    nvs_flash
    backoffAlgorithm
    ota-for-aws-iot-embedded-sdk
    )
#
# Anything that must be exposed to the sources files, but may remain hidden from the header files.
set(PRIV_REQUIRES
    cbor
    coreHTTP
    coreJSON
    posix_compat
    device-shadow-for-aws-iot-embedded-sdk
)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS ${COMPONENT_ADD_INCLUDEDIRS}
                       REQUIRES ${REQUIRES}

(the declaration for SOURCES is not shown here)

Once you have the structure of your project correct, and you apply the "requires" items, you then include the header files that your source needs as normal.

You should be able to avoid using    set(EXTRA_COMPONENT_DIRS     directive as all your components are naturally all in the default areas that are by design.

Look at project examples and dig into the CMake docs a bit to get up to a minimum understanding and you'll find that everything will just work nicely.

tobyloki, 2023-03-08:
I am facing the same issue too. My folder structure is as follows.

```
components/
    esp-aws-iot/
main/
    CMakeLists.txt
CMakeLists.txt
```

The root CMakeLists.txt is as follows.

```
cmake_minimum_required(VERSION 3.5)

set(AWS_IOT_PATH "components/esp-aws-iot")

set(EXTRA_COMPONENT_DIRS
    "${AWS_IOT_PATH}/libraries"
)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project("MyProject")
```

My main CMakeLists.txt is as follows.

```
set(MAIN_REQUIRES
    wifi_provisioning
    coreMQTT
    coreMQTT-Agent
    corePKCS11
    coreJSON
    backoffAlgorithm
    ota-for-aws-iot-embedded-sdk
    Device-Shadow-for-AWS-IoT-embedded-sdk
    unity
)

idf_component_register(
    SRCS
    "main.c"
    INCLUDE_DIRS
    "main"
    REQUIRES
    ${MAIN_REQUIRES}
)

target_add_binary_data(${COMPONENT_TARGET} "certs/aws-root-ca.pem" TEXT)
```

These are my errors.
```
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:275:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xMutex;                    /**< @brief Mutex that protects write operations to the xObjects array. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:288:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xSessionMutex;     /**< @brief Mutex that protects write operations to the pxSession array. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:306:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xVerifyMutex;      /**< @brief Protects the verification key from being modified while in use. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:310:5: error: unknown type name 'mbedtls_threading_mutex_t'
     mbedtls_threading_mutex_t xSignMutex;        /**< @brief Protects the signing key from being modified while in use. */
     ^~~~~~~~~~~~~~~~~~~~~~~~~
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c: In function 'prvMbedTLS_Initialize':
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:459:5: error: implicit declaration of function 'mbedtls_mutex_init'; did you mean 'mbedtls_mpi_init'? [-Werror=implicit-function-declaration]
     mbedtls_mutex_init( &xP11Context.xObjectList.xMutex );
     ^~~~~~~~~~~~~~~~~~
     mbedtls_mpi_init
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c: In function 'prvDeleteObjectFromList':
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:1148:21: error: implicit declaration of function 'mbedtls_mutex_lock'; did you mean 'mbedtls_calloc'? [-Werror=implicit-function-declaration]
     lGotSemaphore = mbedtls_mutex_lock( &xP11Context.xObjectList.xMutex );
                     ^~~~~~~~~~~~~~~~~~
                     mbedtls_calloc
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:1162:18: error: implicit declaration of function 'mbedtls_mutex_unlock'; did you mean 'mbedtls_md_clone'? [-Werror=implicit-function-declaration]
         ( void ) mbedtls_mutex_unlock( &xP11Context.xObjectList.xMutex );
                  ^~~~~~~~~~~~~~~~~~~~
                  mbedtls_md_clone
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c: In function 'C_Finalize':
components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:1490:9: error: implicit declaration of function 'mbedtls_mutex_free'; did you mean 'mbedtls_mpi_free'? [-Werror=implicit-function-declaration]
         mbedtls_mutex_free( &xP11Context.xObjectList.xMutex );
         ^~~~~~~~~~~~~~~~~~
         mbedtls_mpi_free
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

I've tried looking at https://github.com/FreeRTOS/iot-reference-esp32c3 for reference when setting up the CMakeLists.txt files, but I'm not sure what's causing this problem. I want to keep this folder structure (I don't want to copy the esp-aws-iot folder to the esp-idf/components directory). How could I fix this problem?

SolidStateLEDLighting, 2023-03-08:
You may have several errors.... one that I see immediately is:

This is not neccessary: 

set(AWS_IOT_PATH "components/esp-aws-iot")

set(EXTRA_COMPONENT_DIRS
    "${AWS_IOT_PATH}/libraries"
)

You just drop the esp-aws-iot into the components directory and it is by default a locatable component.

Look one of my current projects:

[image]

SolidStateLEDLighting, 2023-03-08:
I have my project set up a bit differently -- my iot component includes these items from the esp-awt-iot component:

[image]

tobyloki, 2023-03-08:
I still had to add the AWS_IOT_PATH to my EXTRA_COMPONENT_DIRS, otherwise it failed. Luckily, I did find a solution and that was to add the following to my sdkconfig.defaults (or sdkconfig) (found from sdkconfig.defaults at https://github.com/FreeRTOS/iot-reference-esp32c3/tree/v202212.00) which resolved the error about mbedtls.

```
CONFIG_MBEDTLS_THREADING_C=y
CONFIG_MBEDTLS_THREADING_ALT=n
CONFIG_MBEDTLS_THREADING_PTHREAD=y
```

I should also note that I'm using the following:

- esp-idf: v4.4.3
- esp-aws-iot: tag 202210.01-LTS-release

SolidStateLEDLighting, 2023-03-08:
You got it.   Good going.  These things are hard.

I'm not sure why you still need to declare the EXTRA_COMPONENT_DIRS.   Something is out of order there.

K.

k-kipruto, 2023-12-17:
> I still had to add the AWS_IOT_PATH to my EXTRA_COMPONENT_DIRS, otherwise it failed. Luckily, I did find a solution and that was to add the following to my sdkconfig.defaults (or sdkconfig) (found [...]

You are a saviour been trying to trouble shoot for days now

SolidStateLEDLighting, 2023-12-17:
It is nice of you to send me some happy feedback.    We are up to v4.4.5 now, but no further.

bilalmalik76, 2023-12-18:
I am also getting the same error and I am using:
* esp-idf: v45.1

* esp-aws-iot: master branch

SolidStateLEDLighting, 2023-12-18:
The new 5.1 is supposed to accept the newest release of the esp-aws-iot library.   They have made a few organizational changes.   I have not tried moving to 5.1 with this library yet.     You best way forward is to sure you understand CMake as best you can and follow any examples you can find.

bilalmalik76, 2023-12-18:
@SolidStateLEDLighting I am abe to resolve mbledtls issues  by following to defconfig: 

CONFIG_MBEDTLS_THREADING_C=y
CONFIG_MBEDTLS_THREADING_ALT=n
CONFIG_MBEDTLS_THREADING_PTHREAD=y

monkeytronics, 2024-01-18:
I encountered the same issue working with 

- ESP-IDF V5.1.2
- esp-aws-iot master branch 

I had to specifically make the changes to config_mbedtls_threading in menuconfig. It wasn't enough to fiddle the values in the sdkconfig.defaults file.

ian-mcbride, 2024-01-19:
Just want to second what @monkeytronics said. Also, shouldn't there be a CMakeLists.txt file in the main esp-aws-iot directory?

SolidStateLEDLighting, 2024-01-19:
Each subdirectory inside the esp-aws-iot library is a component and each has their own CMakeLists.txt files.   You should be able to clone that entire library without modification and just use it.

However, you will need to add those libraries as EXTRA_COMPONENT_DIRS in the root CMakeLists.txt file.

[image]