espressif/esp-aws-iot #169: Have OTA agent and MQTT sharing the same context

raulgotor, 2023-02-08:
Hello, I was running through the `mqtt/tls_mutual_auth` and `ota_mqtt` examples. I'd like to have an application with an MQTT client that is doing the business logic, and at the same time, the OTA agent running on the background (preferably sharing the same context), that would be triggered once a new job is available. 

I started to modify the OTA example, and added some additional MQTT subscriptions and publishes here and there.

For instance, I put some publishing function [here](https://github.com/espressif/esp-aws-iot/blob/1fc7681778bc271960a4e3db514a209df0380917/examples/ota/ota_mqtt/main/ota_demo_core_mqtt.c#L1697), specifically  but then I get errors such as:
```
E (47397) tls_freertos: Reading failed, errno= 128
or
E (48517) esp-tls-mbedtls: read error :-0x004C:
or
E (48527) TRANSPORT_BASE: esp_tls_conn_read error, errno=Socket is not connected
```

On the other hand, having a subscription on that line (guarded for just being executed once) will go ok, but publishing to that topic from AWS MQTT test client at the console won't trigger the callback.

Could you provide some high level implementation guidelines in order to successfully use MQTT once the OTA agent is running in this example?

dhavalgujar, 2023-02-08:
Hi, the coreMQTT-Agent library is designed to solve the problem that you are describing, and is available in `esp-aws-iot`.
Please read my comment [here](https://github.com/espressif/esp-aws-iot/issues/134#issuecomment-1225438158) on #134 for more details and example usage.

raulgotor, 2023-02-11:
This made my day, thanks!