espressif/esp-aws-iot #171: Bootloader verification of AWS IoT OTA signature

projectgus, 2023-03-06:
There's a [mysterious comment in the esp32 PAL for OTA](https://github.com/espressif/esp-aws-iot/blob/master/libraries/ota-for-aws-iot-embedded-sdk/port/ota_pal.c#L509):

```c
/* Write ASN1 decoded signature at the end of firmware image for bootloader to validate during bootup */
```

Is code to do this boot-time verification published anywhere?

Thanks! :grin:

avsheth, 2023-03-08:
Hi @projectgus ,

AWS OTA uses ECDSA for signing the firmware and signature is encoded in ASN1 format. We are extracting this signature and appending at the end of firmware for Espressif bootloader to verify.

In case of secure boot v1, we suggest to upload unsigned firmware to AWS in order to utilise AWS's feature of signing the image using ECDSA. 
For secure boot v2, however, we advise uploading signed firmwares since secure boot v2 uses RSA-PSS for signing and 

AWS OTA service will append the signature (in ASN1 encoded format) regardless, since it can't distinguish between secure boot v1 and v2 firmwares. We have an internal MR which does the signature decoding and appending only for secure boot v1.

The ECDSA bootloader verification snippet would be [this one](https://github.com/espressif/esp-idf/blob/release/v5.0/components/bootloader_support/src/secure_boot_v1/secure_boot_signatures_app.c#L56).

Thanks.

projectgus, 2023-03-08:
Thanks @avsheth for the information. I should have realised this was the Secure Boot V1 format that is written about here, that's very useful.

Unfortunately the project I'm asking on behalf of is Secure Boot V2. Thanks very much for the explanation.

closed 2023-03-08 by projectgus