espressif/esp-aws-iot #173: Compile error

ahmetbilaly, 2023-03-19:
Hi, i am new with esp32 microcontrollers. I am using esp-idf v4.4.4 in vscode. I cloned esp-aws-iot in my project under components folder ( git clone -b "release/202210.01-LTS" --recursive https://github.com/espressif/esp-aws-iot) and I added this line in my root cmake file set(EXTRA_COMPONENT_DIRS "${PROJECT_PATH}/components/esp-aws-iot/libraries") but when i #include “core_mqtt.h” in main.c i took error. What did I wrong?

SolidStateLEDLighting, 2023-03-20:
Add the name of the actual library in that path like this:

"${CMAKE_CURRENT_LIST_DIR}/components/esp-aws-iot/libraries/coreMQTT"

(it is ok to use PROJECT_PATH instead of CMAKE_CURRENT_LIST_DIR)

Then, inside the main folder -- you have another CMakeLlists.txt file -- that file will need to require    coreMQTT

At that point, your "core_mqtt.h" should be reachable.

To understand all this, you will need to read up on CMake and look at good examples and match the patterns of those examples.

K.

ahmetbilaly, 2023-03-21:
@SolidStateLEDLighting thank you so much for quick reply. I continue to research and learn. If I have more questions in the future, can I ask you via e-mail? I promise I won't bother you too much

SolidStateLEDLighting, 2023-03-22:
May I suggest that you use the Esp32 forums also....

closed 2024-01-22 by ahmetbilaly