espressif/esp-aws-iot #175: Help me to choose the correct branch for ESP-IDF v4.4

filgra96, 2023-03-29:
Hi! I'm a newbie for `esp-aws-iot` and I'm trying to understand which branch I should use with my ESP-IDF version.

I'm using the **v4.4**. There are currently 3 branches available for AWS:
- `release/202012.04-LTS` supports the `release/v4.3` and `release/v4.4` versions of ESP-IDF, but it doesn't support `release/v5.0`
- `release/202210.01-LTS` supports the `release/v4.3`, `release/v4.4` and  `release/v5.0` versions of ESP-IDF
- `release/v3.1.x` supports ESP-IDF v3.1 **and above**

I can't understand what should I do.
In a tutorial I saw it does a checkout to `release/v3.1.x` because this branch has the most recent updates *(infact it has the most recent commits)*.
However the `master` branch is located in `release/202210.01-LTS`.

Thank you

dhavalgujar, 2023-03-29:
Hi @filgra96,
[`release/202210.01-LTS`](https://github.com/espressif/esp-aws-iot/tree/release/202210.01-LTS) is the latest release of `esp-aws-iot` and is based on FreeRTOS [202210.01-LTS release](https://github.com/FreeRTOS/FreeRTOS-LTS/releases/tag/202210.01-LTS).
It is compatible with IDF v4.4, making it the right choice for you.

Regarding older releases:
`release/202012.04-LTS` is based on the older FreeRTOS [202012.04-LTS release](https://github.com/FreeRTOS/FreeRTOS-LTS/releases/tag/202012.04-LTS).
`release/v3.1.x` is based on the older (3.x.x) [`aws-iot-device-sdk-embedded-C`](https://github.com/aws/aws-iot-device-sdk-embedded-C/) release.

filgra96, 2023-03-29:
Thank you for your help @dhavalgujar, I don't think I need any further clarification.

closed 2023-03-29 by filgra96