espressif/esp-aws-iot #177: MQTT OTA problem returning FAILED to OTA JOB after starting with new FW.

imammy-hacomono, 2023-04-10:
I tried running the MQTT OTA sample using the Cellular interface Library. 
When I executed the job, block transfer started using MQTT communication. 
After receiving all the blocks, the device restarted, and I confirmed that the app version had been updated to a new one. However, for the OTA job, it published status: FAILED, and it also showed as failed on the AWS console.

I suspect that the cause of this issue is that the values of aflag and pflag read from _esp_get_otadata_partition are both 0xffffffff. 
However, I am just running the sample as is. 
Do I need any additional implementation?

~~~
I (311627) [MQTT OTA]: Sent PUBLISH message: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x00010002"}} 
I (311657) [MQTT OTA]: Received OtaJobEventActivate callback from OTA Agent.
I (311667) esp_image: segment 0: paddr=00110020 vaddr=3f400020 size=27388h (160648) map
I (311717) esp_image: segment 1: paddr=001373b0 vaddr=3ffb0000 size=021e4h (  8676) 
I (311717) esp_image: segment 2: paddr=0013959c vaddr=40080000 size=06a7ch ( 27260) 
I (311737) esp_image: segment 3: paddr=00140020 vaddr=400d0020 size=4ba44h (309828) map
I (311827) esp_image: segment 4: paddr=0018ba6c vaddr=40086a7c size=07840h ( 30784) 
I (311837) esp_image: segment 0: paddr=00110020 vaddr=3f400020 size=27388h (160648) map
I (311887) esp_image: segment 1: paddr=001373b0 vaddr=3ffb0000 size=021e4h (  8676) 
I (311887) esp_image: segment 2: paddr=0013959c vaddr=40080000 size=06a7ch ( 27260) 
I (311897) esp_image: segment 3: paddr=00140020 vaddr=400d0020 size=4ba44h (309828) map
I (311987) esp_image: segment 4: paddr=0018ba6c vaddr=40086a7c size=07840h ( 30784) 
.
.
.
I (312187) [MQTT OTA]:  Received: 132   Queued: 132   Processed: 132   Dropped: 0
.
.
.
I (312447) [MQTT Sub Manager]: Invoking subscription callback of matching topic filter: TopicFilter=$aws/things/+/jobs/#, TopicName=$aws/things/test-things/jobs/AFR_OTA-ota_test/update/accepted
.
.
.
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15500
ho 0 tail 12 room 4
load:0x40080400,len:4072
0x40080400: _init at ??:?

entry 0x40080670
I (29) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (29) boot: compile time 17:05:26
I (29) boot: chip revision: v3.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (76) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 factory          factory app      00 00 00010000 00100000
I (91) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (99) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (106) boot:  6 storage          Unknown data     01 82 00310000 000f0000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=00110020 vaddr=3f400020 size=27388h (160648) map
I (185) esp_image: segment 1: paddr=001373b0 vaddr=3ffb0000 size=021e4h (  8676) load
I (188) esp_image: segment 2: paddr=0013959c vaddr=40080000 size=06a7ch ( 27260) load
I (202) esp_image: segment 3: paddr=00140020 vaddr=400d0020 size=4ba44h (309828) map
I (314) esp_image: segment 4: paddr=0018ba6c vaddr=40086a7c size=07840h ( 30784) load
I (334) boot: Loaded app from partition at offset 0x110000
I (334) boot: Disabling RNG early entropy source...
I (346) cpu_start: Pro cpu up.
I (346) cpu_start: Starting app cpu, entry point is 0x400813f4
0x400813f4: call_start_cpu1 at /Users/keigo.imaizumi/esp/esp-idf/components/esp_system/port/cpu_start.c:142
[... 82 lines cut ...]
I (48272) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0
I (48272) [MQTT OTA]: Received OtaJobEventProcessed callback from OTA Agent.
I (48282) AWS_OTA: Current State=[CreatingFile], Event=[ReceivedJobDocument], New state=[CreatingFile]
I (48292) AWS_OTA: Beginning self-test.
I (48302) ota_pal: otaPal_GetPlatformImageState
I (48302) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (48362) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0
W (48372) AWS_OTA: Rejecting new image and rebooting:The job is in the self-test state while the platform is not.
I (48382) ota_pal: otaPal_SetPlatformImageState, 3
W (48382) ota_pal: Set image as invalid!
I (48392) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (48432) esp_ota_ops: gen_0_seq:1, gen_1_seq:0
I (48442) esp_ota_ops: find_partition->address:d000
I (48442) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0
W (48452) ota_pal: Image not in self test mode 4294967295
I (48462) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (48522) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0
E (48522) AWS_OTA: Job Status Other:1

I (49942) [MQTT Sub Manager]: Invoking subscription callback of matching topic filter: TopicFilter=$aws/things/+/jobs/#, TopicName=$aws/things/test-things/jobs/AFR_OTA-ota_test/update/accepted
E (49962) [MQTT OTA]: prvMqttJobCallback
W (49962) [MQTT OTA]: Received job message $aws/things/test-things/jobs/AFR_OTA-ota_test/update/accepted{"timestamp":1681131347}estamp":1681131340,"execution":{"jobId":"AFR_OTA-ota_test","status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x00010002"},"queuedAt":1681130575,"startedAt":1681130676,"lastUpdatedAt":1681131298,"versionNumber":9,"executionNumber":1,"jobDocument":{"afr_ota":{"protocols":["MQTT"],"streamname":"AFR_OTA-123456-7890-1234-5678-e702a3be8e50","files":[{"filepath":"/update","filesize":537296,"fileid":0,"certfile":"/OTA_Cert/auth.pem","sig-sha256-ecdsa":"aaaaaaaaaaa+bbbbbbbbb+cccccccccccccccccccc/ddddddddddddddddd="}]}}}} size 24.
I (50022) [MQTT OTA]: Sent PUBLISH packet to broker $aws/things/test-things/jobs/AFR_OTA-ota_test/update to broker.
I (50032) [MQTT OTA]: Sent PUBLISH message: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x00000011"}} 
~~~

imammy-hacomono, 2023-04-10:
@dhavalgujar 
Do you know of any solution, or could you please tell me someone who does?
Thank you...

SolidStateLEDLighting, 2023-04-10:
I don't use this OTA agent code -- but I suspect that the self test is failing.   You'll need to dig down into the routines and examine what the self test is trying to verify.  Typically this is something like taking control of a GPIO and toggling it and looking at the output to verify that the new software is in fact working.    Everyone's hardware can be different, so make sure the standard self-test is in fact really working correctly for your hardware.   If not, you'll always be failing and the system will always want to do a roll back.   I write my own self-test in the main and immediately make that determination for the OTA process.   This is what is typically seen in the OTA examples.

imammy-hacomono, 2023-04-11:
@SolidStateLEDLighting 
First of all, thank you for your response!
 It appears that inSelfTestHandler is failing, and it is returning false in the platformInSelftest() function within this function.

~~~
static OtaErr_t inSelfTestHandler(const OtaEventData_t* pEventData) {
    OtaErr_t err = OtaErrNone;

    (void) pEventData;

    LogInfo(("Beginning self-test."));

    /* Check the platform's OTA update image state. It should also be in self test. */
    if (platformInSelftest() == true) {
        /* Callback for application specific self-test. */
        callOtaCallback(OtaJobEventStartTest, NULL);

        /* Clear self-test flag. */
        otaAgent.fileContext.isInSelfTest = false;

        /* Stop the self test timer as it is no longer required. */
        (void) otaAgent.pOtaInterface->os.timer.stop(OtaSelfTestTimer);
    } else {
        /* The job is in self test but the platform image state is not so it could be
         * an attack on the platform image state. Reject the update (this should also
         * cause the image to be erased), aborting the job and reset the device. */
        LogWarn(("Rejecting new image and rebooting:"
                 "The job is in the self-test state while the platform is not."));

        err = setImageStateWithReason(OtaImageStateRejected, (uint32_t) OtaErrImageStateMismatch);
        (void) otaAgent.pOtaInterface->pal.reset(&(otaAgent.fileContext));
    }

    if (err != OtaErrNone) {
        LogError(("Failed to start self-test: "
                  "OtaErr_t=%s",
                  OTA_Err_strerror(err)));
    }

    return err;
}
~~~

In platformInSelftest(), it seems to be checking whether the State is "OtaPalImageStatePendingCommit". Since the State is not "OtaPalImageStatePendingCommit" and it is returning false, it seems that the self-test is not even starting and is failing.

Do you need to implement changing the State to "OtaPalImageStatePendingCommit" yourself? Does anyone know how to do this?
~~~
static bool platformInSelftest(void) {
    bool selfTest = false;

    /*
     * Get the platform state from the OTA pal layer.
     */
    if (otaAgent.pOtaInterface->pal.getPlatformImageState(&(otaAgent.fileContext)) == OtaPalImageStatePendingCommit) {
        selfTest = true;
    }

    return selfTest;
}
~~~

imammy-hacomono, 2023-04-11:
I have submitted an issue to ota-for-aws-iot-embedded-sdk.
Hopefully I can get some answers that will shed some light...

SolidStateLEDLighting, 2023-04-11:
pal stands for Platform Abstraction Layer.    I suggest you follow that function call  getPlatformImageState down until you understand how that routine is going to return OtaPalImageStatePendingCommit.

Seeing as how all hardware can be different, apparently the standard hardware testing procedure doesn't match your hardware?

The ota-for-aws-iot-embedded-sdk is provided by AWS not Espressif.    I would not put too much hope in them because Espressif is breaking away from AWS this is why you are here at esp-aws-iot rather than at aws-iot-embedded-device-sdk-embedded-c.

I was first introduced to the aws-iot-embedded-device-sdk-embedded-c about 2 years back (when Espressif didn't have an answer for AWS) -- and then over time had to discover for myself through indirect observation that Espressif wasn't endorsing that anymore -- they took what they wanted from AWS to produce the aws-iot-sdk.

I have done almost everything possible in aws iot (MQTT, Provisioning, HTTP Rest, Shadow, Jobs, OTA) -- all with the aws-iot-sdk.   If you examine how the sample projects work, you can simplify things quite a bit.   The AWS sdks include so much for every hardware platform (other than esp) that it's a bit daunting to absorb.

This is no cake walk -- you'll need to put in some significant effort.

SolidStateLEDLighting, 2023-04-11:
Notice that there are almost no comments in the code -- you can see what he is doing, but you'll never know why he is doing it.  There is no strategy being described in this file.   To me, this is a major disappointment.   I don't consider this good engineering.

Testing a new OTA download is about making the circuit function to determine if the code is running.  Examples in this library show how pins were being pulled up/down and then examine to verify that code was reading the pins correctly.  In this manor -- all circuits are different -- and if your pins are held low or high -- the canned default test may be failing.  You'll need to find the testing code and see what pins are being manipulated for the test.

SolidStateLEDLighting, 2023-04-12:
I have already given you the "big picture" view.   I don't think you understand it.

imammy-hacomono, 2023-04-12:
We found the cause.
It was because CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE was not enabled.
I had thought it was just a setting to enable rollback, so I had disabled it first,
But I see that by enabling it, the function to update the ota_state flag in the boot loader is also enabled.
I learned a great deal.
https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/system/ota.html#app-rollback

SolidStateLEDLighting, 2023-04-12:
I have not dealt with that before.   I'm not sure why someone would want to disable a roll-back, but there must be a reason for it.

I cut out all that code to keep everything simple.   In my build CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE  is not set.   And my firmware doesn't look for it.

Without extensive documentation and understanding -- all these little things become land-mines.

Sorry that my advice wasn't more helpful to you.

K.

https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/system/ota.html#app-rollback

imammy-hacomono, 2023-04-12:
Every little thing is a minefield. This is so true.
Let's hope this ISUSE is useful to someone!
Thanks for your cooperation!

closed 2023-04-12 by imammy-hacomono