espressif/esp-aws-iot #183: mbedtls_x509_crt_parse returned -0x2180

zhichunlee, 2023-05-22:
Hi
I used thing_shadow example of esp-aws-iot-202210.01-LTS-release to connect AWS IoT, and got follows error info:
 esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180
 esp-tls-mbedtls: Failed to set client pki context
 esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)

The Root CA certificate and client key are downloaded from AWS and worked in old version SDK(perhapse 202203).
Why can't these certificates and keys work with the latest SDK？ What are the differences in certificate and Key management between the latest SDK and the previous version？How do I apply the certificate and key obtained from the SDK of an earlier version to the latest version？

aidiaz, 2023-09-12:
Bumped this to newer release and having the same behaviour. Tested certificates with older project and working just fine :(

zhichunlee, 2023-09-13:
hi Stencht
Unfortunately, I haven't found a solution yet, so I can only use the old version first.

Aldo-GomezCL, 2023-10-09:
Having the exact same problem, good to know that it was already reported, but concerning that the issue is not being looked at.

aidiaz, 2023-10-09:
This issue is due the absence of certificate length definitions, now the length must passed afterwards the certificate in the network contex structure. Somewhere in the examples a certificate is defined in the code and its length passed to the netwrok context.

monkeytronics, 2024-01-26:
@aidiaz I'm having exact same issue. And have already tried adding strlen(cert) and strlen(cert) + 1 as I noticed that the null termination appears to be missing when I retrieve the cert from NVS as always done in previous versions with no issues. Will continue to look into it. I assume someone has figured it out since this issue is months old. If so, would you mind posting your findings?

amitabhskrishworks, 2024-04-03:
I am using the 202210.01-LTS version and facing the same errors, when trying to run the examples/mqtt/tls_mutual_auth example.
Anyone got any kind of solution?
Can anyone of you share the old version which has everything running smoothly?

zhangkehan, 2024-06-03:
I'm having the same problem when I'm running the examples/mqtt/mqtt_demo_mutual_auth under the master branch.
I actually had this problem last year with esp32-c3 but it can run in the version 202012.04-LTS.
But since versions 202012.04-LTS do not support esp32-s3 with IDF v5.1,now I had to use the master branch.
So the problem remains unsolved?

AllanOricil, 2024-12-29:
Im reading the cert.pem and key.pem from an SD Card and I'm having the same problem.

[image]

This is how I'm starting the server

[image]

AllanOricil, 2024-12-29:
I fixed my issue like this

[image]

AllanOricil, 2024-12-30:
My other issue was caused by lack of memory. I can't run https alongside lvgl in my board because there is no enough memory, unfortunately.