espressif/esp-aws-iot #184: Libraries Issue

srk517, 2023-05-25:
Hello I have been trying for days to use this repository. There are many libraries that are included in code but are not in package itself. Even if I comment those libraries there are some unknown variables that are not declared, because that basic library is missing. I don not understand what to do now. Please if you make a repository especially for esp-idf AWS it has to work. We have business to run and this is causing delay! Libraries like #inlcude core_mqtt_state.h is not in packate , core_mqtt.h is not in package. Please help!!

idea--list, 2023-05-25:
@srk517 
I completely understand your situation, i also was in this phase. IMHO this repo is rather a proof of concept and i would not use it production as it is not meeting that standard of quality. To make it worse even the "production-ready" [reference example](https://github.com/FreeRTOS/iot-reference-esp32c3) is also not quite ready for production.

I find both these repos suffer from design issues (like each demo task has its own event handler functions but with the same names, etc.). When trying to run all 3 examples the same time, integrated into a project that was stable before, my ESP always hangs within 1-2 hours without giving me any useful info for debugging.

So IMHO you have to study&dissect both repos and then implement your own version for production. Wish this was not the case though.

srk517, 2023-05-26:
Thanks for reply. This is my first time integrating esp-idf with aws. I am working with idf mqtt ssl_mutual_auth example now it is far more better than this aws repo. I think i should keep on exploring the example part instead of this because it has wasted so much of my time on esp_aws_iot repo. Hopefully that would work otherwise i need to shift to Arduino IDE.

SolidStateLEDLighting, 2023-06-07:
The library is working for me -- but I took the time to completely rewrite all the examples into my own implementation.   This is a very long and time-consuming process.   The results that I have are good for the library calls.

One key point is -- that you need to make sure your project is very component based because the library will drop into your project as a component.   

The next most important thing is probably to realize that MQTT_ProcessLoop() must be written into your program flow and circulated through at all times --- when you have something to do and when you are just at idle.

Use state modeling techniques to move from one activity to another.

See the attached snapshot.   I included the aws libraries without the example code -- then I wrote my own IOT component to call those libraries. 

[image]

srk517, 2023-06-08:
Hello Keith , Thank you for replying. I have fortunately completed my task but used SSL Mutual Auth example. I did not follow this repository or any form of aws ready made. But I would love to learn more. Thanks!

srk517, 2023-06-08:
Issue resolved by use SSL Mutual Auth example