espressif/esp-aws-iot #189: No response from AWS

schaffster, 2023-06-13:
Pkcs setup successful,  no response from AWS connect.     I'm using esp-idf 4.4.3.

Does this code require a specific esp32 chip?

Not using c3

GlenSchaff, 2023-06-22:
Hi @avsheth 

I make it up to the MQTT_connect function in my code, during MQTT _connect() I start getting the transport Errors directly below.  Below that is my code followed by a full connection log.

Thanks

E (31993) Transport_MbedTLS_PKCS11: Failed to read data: mbedTLSError= SSL - The peer notified us that the connection is going to be closed : <No-Low-Level-Code>.
E (32003) coreMQTT: A single byte was not read from the transport: transportStatus=-30848.
E (32013) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed.
E (32013) coreMQTT: MQTT connection failed with status = MQTTRecvFailed.
E (32023) EstablishMqttSession: Connection with MQTT broker failed with status MQTTRecvFailed
E (32033) aws_fleet_provisioning_task: Failed to establish MQTT session
Thanks

    ESP_LOGI(__FUNCTION__,"Client ID:%s",uSwitch.Net.sHostName);
                /* The maximum time interval in seconds which is allowed to elapse
                 * between two Control Packets.
                 * It is the responsibility of the client to ensure that the interval between
                 * control packets being sent does not exceed the this keep-alive value. In the
                 * absence of sending any other control packets, the client MUST send a
                 * PINGREQ packet. */
                connectInfo.keepAliveSeconds=MQTT_KEEP_ALIVE_INTERVAL_SECONDS;
                /* Username and password for authentication. Not used in this demo. */
                connectInfo.pUserName=MQTT_METRICS_STRING;
                connectInfo.userNameLength=strlen(MQTT_METRICS_STRING);
                connectInfo.pPassword=NULL;
                connectInfo.passwordLength=0U;
                /* Send an MQTT CONNECT packet to the broker. */
                mqttStatus=MQTT_Connect(pMqttContext,&connectInfo,NULL,MQTT_CONNACK_RECV_TIMEOUT_MS,&sessionPresent);
                if(mqttStatus != MQTTSuccess){
                    ret=false;
                    ESP_LOGE(__FUNCTION__,"Connection with MQTT broker failed with status %s",MQTT_Status_strerror(mqttStatus));
                }else{
                    ESP_LOGI(__FUNCTION__,"MQTT Connection Successful");
                }
            }
        }

W (5823) wifi:<ba-add>idx:1 (ifx:0, c8:9e:43:76:d8:53), tid:0, ssn:0, winSize:64
I (23693) corePKCS11: PKCS #11 successfully initialized.
I (23703) loadClaimCredentials: Provision Private Key
I (23713) corePKCS11: Creating a 0x3 type object.
I (23713) PKCS11: Initializing NVS partition: "storage"
I (23723) loadClaimCredentials: Provision Certificate
E (23733) corePKCS11: Failed to destroy object. PKCS11_PAL_DestroyObject failed.
I (23733) provisionCertificate: Writing certificate into label "Claim Cert".
I (23743) corePKCS11: Creating a 0x1 type object.
I (23763) aws_fleet_provisioning_task: Establishing MQTT session with claim certificate
I (23763) connectToBrokerWithBackoffRetries: Establishing a TLS session to ag3lxmk29btzg-ats.iot.us-east-1.amazonaws.com:8883
I (31863) connectToBrokerWithBackoffRetries: Connected to Broker using PKCS
I (31863) EstablishMqttSession: Client ID:USwitch_C049EF926225
E (31993) Transport_MbedTLS_PKCS11: Failed to read data: mbedTLSError= SSL - The peer notified us that the connection is going to be closed : <No-Low-Level-Code>.
E (32003) coreMQTT: A single byte was not read from the transport: transportStatus=-30848.
E (32013) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed.
E (32013) coreMQTT: MQTT connection failed with status = MQTTRecvFailed.
E (32023) EstablishMqttSession: Connection with MQTT broker failed with status MQTTRecvFailed
E (32033) aws_fleet_provisioning_task: Failed to establish MQTT session
I (32043) corePKCS11: Successfully closed PKCS #11 session.
I (32043) corePKCS11: PKCS #11 was successfully uninitialized.
I (32053) aws_fleet_provisioning_task: Fleet Provisioning f

mdjavedakhtar, 2025-01-16:
is this resolved ??