espressif/esp-aws-iot #191: "Factory" partition and "otadata partition is invalid, factory/ota_0 is boot partition" - OTA fails

iandaintith, 2023-07-10:
**Platform : ESP32 

Branch :  https://github.com/espressif/esp-aws-iot/tree/release/202210.01-LTS

ESP_IDF : esp-idf-v4.4.4**

Problem :  
OTA Agent fails to boot the downloaded image if partition csv contains 'factory' partition. If 'factory' partition is removed from csv then the OTA agent works OK.

Partition table that FAILS

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     0x9000,   16k,
otadata,  data, ota,     0xd000,   8K,
phy_init, data, phy,     0xf000,   4k,
factory,  app,  factory, 0x10000,  2M,
ota_0,    app,  ota_0,   0x210000, 2M,
ota_1,    app,  ota_1,   0x410000, 2M,
storage,  data, spiffs,          , 1M,
```

Partition table that SUCCEEDS

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap,,,,
nvs,      data, nvs,     0x9000,   16k,
otadata,  data, ota,     0xd000,   8K,
phy_init, data, phy,     0xf000,   4k,
ota_0,    app,  ota_0,   0x210000, 2M,
ota_1,    app,  ota_1,   0x410000, 2M,
storage,  data, spiffs,          , 1M,

```

**Debug**

On initialising the OTA agent we get this warning 
```
I (49719) mqtt-agent: establishing a TLS session to a14vwue44pf12r-ats.iot.eu-west-2.amazonaws.com:8883 ...
I (51179) mqtt-agent: establishing a MQTT connection for client: 78218443E4AC ...
I (51579) coreMQTT: MQTT connection established with the broker.
I (51579) mqtt-agent: session present: 0
I (51579) mqtt-agent: connected to AWS endpoint
I (52739) zappi: set rssi:-72
I (53139) AWS_OTA: otaPal_GetPlatformImageState
**I (53139) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
W (53149) esp_ota_ops: otadata partition is invalid, factory/ota_0 is boot partition**
```

This warning is traced to line 77 of aws_esp_ota_ops.c in _esp_get_otadata_partition(). 

Call stack appears to be _esp_get_otadata_partition() -> ota_select_valid() -> 

On the reboot after the download 

```
 (21094) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (21094) coreMQTT: State record updated. New state=MQTTPublishDone.
I (21094) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (21104) coreMQTT: State record updated. New state=MQTTPublishDone.
I (21114) mqtt-agent: incoming publish passed to subscription manager
I (21104) mqtt-ota: sent PUBLISH packet to broker $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V003-3/update to broker
I (21124) AWS_OTA: Job parsing success: OtaJobParseErr_t=OtaJobParseErrNone, Job name=AFR_OTA-78218443E4AC-V003-3
I (21144) AWS_OTA: otaPal_GetPlatformImageState
**I (21144) esp_ota_ops: aws_esp_ota_get_boot_flags: 1**
**I (21174) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0**
I (21174) mqtt-ota: received OtaJobEventProcessed callback from OTA Agent.
I (21174) AWS_OTA: Current State=[CreatingFile], Event=[ReceivedJobDocument], New state=[CreatingFile]
I (21184) AWS_OTA: Beginning self-test.
I (21184) AWS_OTA: otaPal_GetPlatformImageState
I (21194) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
**I (21214) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0**
W (21214) AWS_OTA: Rejecting new image and rebooting:The job is in the self-test state while the platform is not.
I (21214) AWS_OTA: otaPal_SetPlatformImageState, 3
W (21224) AWS_OTA: Set image as invalid!
I (21224) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (21254) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0
W (21254) AWS_OTA: Image not in self test mode 4294967295
I (21254) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (21264) esp_ota_ops: [0] aflags/seq:0xffffffff/0x1, pflags/seq:0xffffffff/0x0
I (21264) mqtt-ota: publishing.... to topic $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V003-3/update
I (22114) coreMQTT: Publishing message to $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V003-3/update.
```

I have seen references to this issue here https://github.com/aws/amazon-freertos/issues/1429 dating to 2019 but was supposedly fixed?

I would be grateful if you would look into this. Thanks.

iandaintith, 2023-07-10:
Further information. When factory partition is removed the boot flags appear valid. 

**Before the OTA job runs** 

I (51397) mqtt-agent: connected to AWS endpoint
I (52737) zappi: set rssi:-78
I (53157) AWS_OTA: otaPal_GetPlatformImageState
I (53157) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (53167) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (53177) AWS_OTA: Current State=[RequestingJob], Event=[Resume], New state=[RequestingJob]
I (53177) mqtt-ota: subscribing.... to topic $aws/things/78218443E4AC/jobs/notify-next with broker
I (54187) mqtt-ota: resume OK
I (54427) mqtt-ota: registering topic $aws/things/78218443E4AC/jobs/notify-next
I (54427) mqtt-ota: [0] isMatch:1 with $aws/things/+/jobs/notify-next
I (54427) mqtt-ota: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/jobs/notify-next
I (54437) mqtt-ota: [1] isMatch:0 with $aws/things/+/streams/#
I (54447) mqtt-ota: [2] isMatch:1 with $aws/things/+/jobs/#
I (54457) mqtt-ota: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/jobs/notify-next
I (54477) mqtt-ota: SUBSCRIBED to topic $aws/things/78218443E4AC/jobs/notify-next to broker
I (54477) AWS_OTA: Subscribed to MQTT topic: $aws/things/78218443E4AC/jobs/notify-next
I (54487) mqtt-ota: publishing.... to topic $aws/things/78218443E4AC/jobs/$next/get
I (55467) coreMQTT: Publishing message to $aws/things/78218443E4AC/jobs/$next/get.

I (56487) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (56487) coreMQTT: State record updated. New state=MQTTPublishDone.
I (56487) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (56497) coreMQTT: State record updated. New state=MQTTPublishDone.
I (56507) mqtt-agent: incoming publish passed to subscription manager
I (56507) mqtt-ota: received job message callback, size 54
I (56507) mqtt-ota: sent PUBLISH packet to broker $aws/things/78218443E4AC/jobs/$next/get to broker
I (56537) AWS_OTA: Current State=[WaitingForJob], Event=[RequestJobDocument], New state=[WaitingForJob]
I (56537) AWS_OTA: No active job available in received job document: OtaJobParseErr_t=OtaJobParseErrNoActiveJobs
I (56547) AWS_OTA: otaPal_GetPlatformImageState
**I (56557) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (56577) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0**
I (56577) mqtt-ota: received OtaJobEventProcessed callback from OTA Agent.
I (56577) AWS_OTA: Current State=[WaitingForJob], Event=[ReceivedJobDocument], New state=[CreatingFile]
I (158517) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (158517) coreMQTT: State record updated. New state=MQTTPubAckSend.
I (158517) mqtt-agent: incoming publish passed to subscription manager
I (158527) mqtt-ota: received job message callback, size 494
I (158547) AWS_OTA: Extracted parameter: [key: value]=[execution.jobId: AFR_OTA-78218443E4AC-V004-2]
I (158547) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr_ota.streamname: AFR_OTA-71165c7f-ec65-4faa-a0ba-dc6a085c6217]
I (158557) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr_ota.protocols: ["MQTT"]]
I (158577) AWS_OTA: Extracted parameter: [key: value]=[filepath: /]
I (158577) AWS_OTA: Extracted parameter: [key: value]=[filesize: 1019952]
I (158587) AWS_OTA: Extracted parameter: [key: value]=[fileid: 0]
I (158587) AWS_OTA: Extracted parameter: [key: value]=[certfile: Code Verify Key]
I (158597) AWS_OTA: Extracted parameter [ sig-sha256-ecdsa: MEUCIGX4QTMzjvOK2IethxyJCiboL+Rm... ]
I (158617) AWS_OTA: Job document was accepted. Attempting to begin the update.
I (158617) AWS_OTA: Job parsing success: OtaJobParseErr_t=OtaJobParseErrNone, Job name=AFR_OTA-78218443E4AC-V004-2
I (158627) AWS_OTA: otaPal_GetPlatformImageState
I (158627) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (158647) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (158647) AWS_OTA: Writing to partition subtype 17 at offset 0x410000

**Last block of OTA** 

I (582127) mqtt-ota: received OtaJobEventProcessed callback from OTA Agent.
I (582137) AWS_OTA: Current State=[WaitingForFileBlock], Event=[ReceivedFileBlock], New state=[WaitingForFileBlock]
I (582157) AWS_OTA: Received valid file block: Block index=993, Size=1024
I (582157) AWS_OTA: Received final block of the update.
I (582737) zappi: set rssi:-79
I (583497) AWS_OTA: Signature verification succeeded.
I (583497) AWS_OTA: Received entire update and validated the signature.
I (583497) mqtt-ota: publishing.... to topic $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V004-2/update
I (583517) coreMQTT: Publishing message to $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V004-2/update.

I (584527) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (584527) coreMQTT: State record updated. New state=MQTTPublishDone.
[... 56 lines cut ...]
I (21091) AWS_OTA: otaPal_GetPlatformImageState
I (21101) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
**I (21121) esp_ota_ops: [1] aflags/seq:0x1/0x2, pflags/seq:0x2/0x1**
I (21121) mqtt-ota: received OtaJobEventStartTest callback from OTA Agent
I (21121) AWS_OTA: otaPal_SetPlatformImageState, 2
**I (21131) AWS_OTA: Set image as valid one!**
I (21131) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
**I (21151) esp_ota_ops: [1] aflags/seq:0x1/0x2, pflags/seq:0x2/0x1**
I (21151) esp_ota_ops: aws_esp_ota_set_boot_flags: 2 1
[... cut ...]
I (22231) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (22241) coreMQTT: State record updated. New state=MQTTPublishDone.
I (22241) mqtt-agent: incoming publish passed to subscription manager
I (22231) mqtt-ota: sent PUBLISH packet to broker $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V004-2/update to broker
I (22281) ota_pal: Running partition type 0 subtype 17 (offset 0x00410000)
I (22281) ota_pal: Configured partition type 0 subtype 17 (offset 0x00410000)
I (22281) ota_pal: Current running firmware is: ota_1

SolidStateLEDLighting, 2023-07-16:
I strongly suspect that you have the wrong offsets in your partition table.   Both tables above can't be correct.

Delete the offsets and allow the tools to calculate them for you.

iandaintith, 2023-07-16:
Thanks but the 2nd partition table works fine. Nothing wrong with offsets.

SolidStateLEDLighting, 2023-07-16:
You inserted a factory partition in table 1 and now the offsets are screwed up.   ota_0 and ota_1 couldn't possibly be at the same offsets as table 2.

SolidStateLEDLighting, 2023-07-16:
Do a full clean on the project and rebuild.

iandaintith, 2023-07-16:
Thanks. Ive done many clean builds with full ota process using table 1 and table 2. I believe the problem is related to the number of bootable partitions and that the library is expecting to see ota1 and ota2 not factory when it enumerates. There was a previously problem in 2019 (see link) which was the same symptom. By means of explanation this is existing code thats been in the field for years upgrading using esp https ota. The only thing I changed was adding in the aws iot ota lib. Thanks for your comments.

SolidStateLEDLighting, 2023-07-16:
I use a table that is slightly more complex than this one and I never have a problem with it -- but I don't call an agent in my AWS IOT project.

The formatting doesn't hold here so the columns are not nice to look at.

# ESP-IDF Partition Table
# Name,     Type,   SubType,    Offset,   Size,      Flags
nvs,        data,   nvs,              ,   0x006000,
nvs_key,    data,   nvs_keys,         ,   0x001000,  encrypted
phy_init,   data,   phy,              ,   0x001000,  encrypted
otadata,    data,   ota,              ,   0x002000,
factory,    app,    factory,          ,   0x271000,
ota_0,      app,    ota_0,            ,   0x67D000,
ota_1,      app,    ota_1,            ,   0x67D000,

iandaintith, 2023-07-16:
Thanks. Ill try this when back in the office.

iandaintith, 2023-07-26:
Hi I tried the the partitions.csv without offsets and received the same error. 

I (53252) AWS_OTA: otaPal_GetPlatformImageState
I (53252) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
**W (53262) esp_ota_ops: otadata partition is invalid, factory/ota_0 is boot partition**

SolidStateLEDLighting, 2023-07-26:
I don't know why they make things so complicated.    Almost everything is subscribe, publish, handle the publish-out response, and unsubscribe.

iandaintith, 2023-07-26:
Despite that warning the OTA succeeded. Just an error erasing the old partition?

I (23578) mqtt-agent: incoming publish passed to subscription manager
I (23578) mqtt-common: sent PUBLISH packet to broker $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V006-1521/update to broker
I (23598) AWS_OTA: Job parsing success: OtaJobParseErr_t=OtaJobParseErrNone, Job name=AFR_OTA-78218443E4AC-V006-1521
I (23608) AWS_OTA: otaPal_GetPlatformImageState
I (23618) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (23628) esp_ota_ops: [0] aflags/seq:0x1/0x1, pflags/seq:0xffffffff/0x0
I (23628) mqtt-ota: received OtaJobEventProcessed callback from OTA Agent.
I (23638) AWS_OTA: Current State=[CreatingFile], Event=[ReceivedJobDocument], New state=[CreatingFile]
I (23648) AWS_OTA: Beginning self-test.
I (23648) AWS_OTA: otaPal_GetPlatformImageState
I (23658) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (23668) esp_ota_ops: [0] aflags/seq:0x1/0x1, pflags/seq:0xffffffff/0x0
I (23668) mqtt-ota: received OtaJobEventStartTest callback from OTA Agent
I (23678) AWS_OTA: otaPal_SetPlatformImageState, 2
**I (23688) AWS_OTA: Set image as valid one!**
I (23688) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (23698) esp_ota_ops: [0] aflags/seq:0x1/0x1, pflags/seq:0xffffffff/0x0
I (23698) esp_ota_ops: aws_esp_ota_set_boot_flags: 2 1
I (23718) esp_ota_ops: [0] aflags/seq:0x1/0x1, pflags/seq:0xffffffff/0x0
I (23778) AWS_OTA: Disabling RTC hardware watchdog timer
I (23778) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (23788) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (23788) mqtt-ota: publishing.... to topic $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V006-1521/update
I (24798) mqtt-common: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected
I (24818) mqtt-common: SUBSCRIBED to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected to broker
I (24818) mqtt-common: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected
I (24828) mqtt-common: subscribing.... to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/update/delta with broker
I (25808) coreMQTT: Publishing message to $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V006-1521/update.

I (26008) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (26008) coreMQTT: State record updated. New state=MQTTPublishDone.
I (26018) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (26018) coreMQTT: State record updated. New state=MQTTPublishDone.
I (26028) mqtt-agent: incoming publish passed to subscription manager
I (26028) mqtt-common: sent PUBLISH packet to broker $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V006-1521/update to broker
I (26058) ota_pal: Running partition type 0 subtype 16 (offset 0x00210000)
I (26058) ota_pal: Configured partition type 0 subtype 16 (offset 0x00210000)
I (26068) ota_pal: Current running firmware is: ota_0
**E (26088) mqtt-ota: failed to erase last boot partition! (-1)**
I (26098) AWS_OTA: Current State=[WaitingForJob], Event=[StartSelfTest], New state=[WaitingForJob]
I (26898) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (26898) coreMQTT: State record updated. New state=MQTTPubAckSend.

SolidStateLEDLighting, 2023-07-26:
I believe that in my process, I erase the target partition just before I receive the new update.  Its a slow process for a large partition.

iandaintith, 2023-07-26:
Just tried a subsequent OTA and it has worked again with no warning.

I think we have cracked it. Thank you. 

I (51191) mqtt-agent: establishing a MQTT connection for client: 78218443E4AC ...
I (51341) coreMQTT: MQTT connection established with the broker.
I (51341) mqtt-agent: session present: 0
I (51341) mqtt-agent: connected to AWS endpoint
I (52791) zappi: set rssi:-66
I (53271) AWS_OTA: otaPal_GetPlatformImageState
I (53271) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (53281) esp_ota_ops: [1] aflags/seq:0x1/0x2, pflags/seq:0x2/0x1
I (53311) AWS_OTA: Current State=[RequestingJob], Event=[Resume], New state=[RequestingJob]
I (53311) mqtt-ota: subscribing.... to topic $aws/things/78218443E4AC/jobs/notify-next with broker
I (54321) mqtt-ota: resume OK
I (54361) mqtt-ota: registering topic $aws/things/78218443E4AC/jobs/notify-next
I (54361) mqtt-ota: [0] isMatch:1 with $aws/things/+/jobs/notify-next
I (54361) mqtt-ota: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/jobs/notify-next
I (54371) mqtt-ota: [1] isMatch:0 with $aws/things/+/streams/#
I (54381) mqtt-ota: [2] isMatch:1 with $aws/things/+/jobs/#
I (54391) mqtt-ota: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/jobs/notify-next
I (54411) mqtt-ota: SUBSCRIBED to topic $aws/things/78218443E4AC/jobs/notify-next to broker
I (54411) AWS_OTA: Subscribed to MQTT topic: $aws/things/78218443E4AC/jobs/notify-next
I (54431) mqtt-ota: publishing.... to topic $aws/things/78218443E4AC/jobs/$next/get
I (54421) mqtt-shadow: MQTT agent connected!
I (54541) mqtt-shadow: subscribing ...
I (54541) mqtt-common: Subscribe all. 5 topics
I (54541) mqtt-common: subscribing.... to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/accepted with broker
I (54871) mqtt-api: MQTT agent connected!
I (54971) mqtt-api: subscribing ...
I (54971) mqtt-common: Subscribe all. 1 topics
I (54971) mqtt-common: subscribing.... to topic 78218443E4AC/hubcommand with broker
I (55401) coreMQTT: Publishing message to $aws/things/78218443E4AC/jobs/$next/get.

I (55461) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (55461) coreMQTT: State record updated. New state=MQTTPublishDone.
I (55471) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (55471) coreMQTT: State record updated. New state=MQTTPublishDone.
I (55481) mqtt-agent: incoming publish passed to subscription manager
I (55491) mqtt-ota: received job message callback, size 619
I (55471) mqtt-common: sent PUBLISH packet to broker $aws/things/78218443E4AC/jobs/$next/get to broker
I (55511) AWS_OTA: Current State=[WaitingForJob], Event=[RequestJobDocument], New state=[WaitingForJob]
I (55511) mqtt-common: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/accepted
I (55531) AWS_OTA: Extracted parameter: [key: value]=[execution.jobId: AFR_OTA-78218443E4AC-V007-1528]
I (55541) AWS_OTA: Extracted parameter: [key: value]=[execution.statusDetails.updatedBy: 6]
I (55551) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr_ota.streamname: AFR_OTA-36573db5-4edf-4664-b811-c319ef1d75ab]
I (55571) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr_ota.protocols: ["MQTT"]]
I (55571) AWS_OTA: Extracted parameter: [key: value]=[filepath: /]
I (55591) AWS_OTA: Extracted parameter: [key: value]=[filesize: 1006864]
I (55541) mqtt-common: SUBSCRIBED to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/accepted to broker
I (55591) AWS_OTA: Extracted parameter: [key: value]=[fileid: 0]
I (55611) mqtt-common: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/accepted
I (55621) mqtt-common: subscribing.... to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected with broker
I (55611) AWS_OTA: Extracted parameter: [key: value]=[certfile: Code Verify Key]
I (55641) AWS_OTA: Extracted parameter [ sig-sha256-ecdsa: MEYCIQDmurizsHkhIZwQvS/oIOt9Ennp... ]
I (55661) AWS_OTA: In self test mode.
I (55661) AWS_OTA: New image has a higher version number than the current image: New image version=0.0.7, Previous image version=0.0.6
I (55671) AWS_OTA: Image version is valid: Begin testing file: File ID=0
I (55681) AWS_OTA: otaPal_SetPlatformImageState, 1
**W (55691) AWS_OTA: Set image as testing!**
I (55691) mqtt-ota: publishing.... to topic $aws/things/78218443E4AC/jobs/AFR_OTA-78218443E4AC-V007-1528/update
I (56601) mqtt-common: registered a publish callback with subscription manager for topic 78218443E4AC/hubcommand
I (56621) mqtt-common: SUBSCRIBED to topic 78218443E4AC/hubcommand to broker
I (56621) mqtt-common: registered a publish callback with subscription manager for topic 78218443E4AC/hubcommand
I (56621) mqtt-api: subscribed OK
I (56791) mqtt-common: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected
I (56811) mqtt-common: SUBSCRIBED to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected to broker
I (56821) mqtt-common: registered a publish callback with subscription manager for topic $aws/things/78218443E4AC/shadow/name/deviceConfig/delete/rejected
I (56831) mqtt-common: subscribing.... to topic $aws/things/78218443E4AC/shadow/name/deviceConfig/update/delta with broker
[... cut ...]
I (66631) AWS_OTA: Current State=[WaitingForJob], Event=[StartSelfTest], New state=[WaitingForJob]
I (66651) AWS_OTA: No active job available in received job document: OtaJobParseErr_t=OtaJobParseErrNoActiveJobs
I (66651) AWS_OTA: otaPal_GetPlatformImageState
I (66661) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (66681) esp_ota_ops: [1] aflags/seq:0x2/0x2, pflags/seq:0xffffffff/0x0
I (66681) mqtt-ota: received OtaJobEventProcessed callback from OTA Agent.
I (66681) AWS_OTA: Current State=[WaitingForJob], Event=[ReceivedJobDocument], New state=[CreatingFile]

iandaintith, 2023-07-26:
Thanks for your help 👍

closed 2023-07-26 by iandaintith