espressif/esp-aws-iot #197: Support ESP-IDF v5.1 and v5.2

Stencht, 2023-08-28:
Hi, 

I was wondering if there will be support for ESP-IDF v5.1 and v5.2?
If so, when is this to be expected?

Currently it is not possible to build the example from `examples/ota/ota_mqtt` using v5.1 or v5.2.
The error I get is: `error: unknown type name 'osi_sem_t'` although it is defined in `semaphore.h`, which is included.

The build output for IDF v5.2 (similar for v5.1):

```
-- Found Git: C:/Users/USER/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/USER/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/USER/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/USER/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependency "idf(5.1.0)" version changed, solving dependencies.
..............Updating lock file at C:\Projects\PROJ\esp-aws-iot\examples\ota\ota_mqtt\dependencies.lock
Processing 3 dependencies:
[1/3] espressif/cbor (0.6.0~1)
[2/3] espressif/esp_secure_cert_mgr (2.3.1)
[3/3] idf (5.2.0)
-- Project sdkconfig file C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/sdkconfig
Loading defaults file C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/sdkconfig.defaults...
C:\Users\USER\AppData\Local\Temp\kconfgen_tmp1nx_xxqf:5 line was updated to CONFIG_NEWLIB_NANO_FORMAT=n
C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/sdkconfig:527 CONFIG_COMPILER_OPTIMIZATION_DEFAULT was replaced with CONFIG_COMPILER_OPTIMIZATION_DEBUG
-- Compiler supported targets: xtensa-esp32s3-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/USER/.espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe (found version "3.11.0") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "ota_mqtt" version: 202210.01-LTS-release
-- Adding linker script C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Users/USER/esp/esp-idf-master/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update backoffAlgorithm bootloader bootloader_support bt cmock console coreJSON coreMQTT corePKCS11 cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cbor espressif__esp_secure_cert_mgr esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread ota-for-aws-iot-embedded-sdk partition_table perfmon posix_compat protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/USER/esp/esp-idf-master/esp-idf/components/app_trace C:/Users/USER/esp/esp-idf-master/esp-idf/components/app_update C:/Projects/PROJ/esp-aws-iot/libraries/backoffAlgorithm C:/Users/USER/esp/esp-idf-master/esp-idf/components/bootloader C:/Users/USER/esp/esp-idf-master/esp-idf/components/bootloader_support C:/Users/USER/esp/esp-idf-master/esp-idf/components/bt C:/Users/USER/esp/esp-idf-master/esp-idf/components/cmock C:/Users/USER/esp/esp-idf-master/esp-idf/components/console C:/Projects/PROJ/esp-aws-iot/libraries/coreJSON C:/Projects/PROJ/esp-aws-iot/libraries/coreMQTT C:/Projects/PROJ/esp-aws-iot/libraries/corePKCS11 C:/Users/USER/esp/esp-idf-master/esp-idf/components/cxx C:/Users/USER/esp/esp-idf-master/esp-idf/components/driver C:/Users/USER/esp/esp-idf-master/esp-idf/components/efuse C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp-tls C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_adc C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_app_format C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_bootloader_format C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_coex C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_common C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_eth C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_event C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_gdbstub C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_hid C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_http_client C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_http_server C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_https_ota C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_https_server C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_hw_support C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_lcd C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_local_ctrl C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_mm C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_netif C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_netif_stack C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_partition C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_phy C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_pm C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_psram C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_ringbuf C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_rom C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_system C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_timer C:/Users/USER/esp/esp-idf-master/esp-idf/components/esp_wifi C:/Users/USER/esp/esp-idf-master/esp-idf/components/espcoredump C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/managed_components/espressif__cbor C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/managed_components/espressif__esp_secure_cert_mgr C:/Users/USER/esp/esp-idf-master/esp-idf/components/esptool_py C:/Users/USER/esp/esp-idf-master/esp-idf/components/fatfs C:/Users/USER/esp/esp-idf-master/esp-idf/components/freertos C:/Users/USER/esp/esp-idf-master/esp-idf/components/hal C:/Users/USER/esp/esp-idf-master/esp-idf/components/heap C:/Users/USER/esp/esp-idf-master/esp-idf/components/http_parser C:/Users/USER/esp/esp-idf-master/esp-idf/components/idf_test C:/Users/USER/esp/esp-idf-master/esp-idf/components/ieee802154 C:/Users/USER/esp/esp-idf-master/esp-idf/components/json C:/Users/USER/esp/esp-idf-master/esp-idf/components/log C:/Users/USER/esp/esp-idf-master/esp-idf/components/lwip C:/Projects/PROJ/esp-aws-iot/examples/ota/ota_mqtt/main C:/Users/USER/esp/esp-idf-master/esp-idf/components/mbedtls C:/Users/USER/esp/esp-idf-master/esp-idf/components/mqtt C:/Users/USER/esp/esp-idf-master/esp-idf/components/newlib C:/Users/USER/esp/esp-idf-master/esp-idf/components/nvs_flash C:/Users/USE [... cut ...]
c:\projects\PROJ\esp-aws-iot\examples\ota\ota_mqtt\managed_components\espressif__cbor\tinycbor\src\cbor.h:437:75: note: expected 'int *' but argument is of type 'int32_t *' {aka 'long int *'}
  437 | CBOR_INLINE_API CborError cbor_value_get_int(const CborValue *value, int *result)
      |                                                                      ~~~~~^~~~~~
[982/1002] Building C object esp-idf/coreMQTT/CMakeFiles/__idf_coreMQTT.dir/coreMQTT/source/core_mqtt.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. 
```

jsanchezerreka, 2023-09-07:
Hi, any news about this? 

I need to use esp-idf 5.1 as it implements the possibility to use secure element in esp_http_client.

naeben, 2023-09-29:
No update from my part, I also just reverted to 5.0 which works fine (as long as you don't need stuff that only works in the later versions ;-) )

Stencht, 2023-09-29:
No update from my part, I also just reverted to 5.0 which works fine (as long as you don't need stuff that only works in the later versions ;-) )

jsanchezerreka, 2023-09-29:
Neither from here. I also reverted to 5.0 meanwhile and as I need some 5.1 features I was forced to manually change that code to activate those features... quite disgusting but it works for the moment :)

dhavalgujar, 2023-10-02:
Hi everyone, this has been fixed in one of the recent commits ([69f9995](https://github.com/espressif/esp-aws-iot/commit/69f9995b1b1d001f4c1e9624695227c1063ab513)) made to the master branch.  
All examples should now build and work correctly on IDF v5.1.X

closed 2023-10-02 by dhavalgujar

OussamaBoualii, 2024-06-13:
> Hi everyone, this has been fixed in one of the recent commits ([69f9995](https://github.com/espressif/esp-aws-iot/commit/69f9995b1b1d001f4c1e9624695227c1063ab513)) made to the master branch. All examples should now build and work correctly on IDF v5.1.X

> Hi everyone, this has been fixed in one of the recent commits ([69f9995](https://github.com/espressif/esp-aws-iot/commit/69f9995b1b1d001f4c1e9624695227c1063ab513)) made to the master branch. All examples should now build and work correctly on IDF v5.1.X

I am still getting same error, idf version 5.2 (similar error 5.1) ota_mqtt example.