espressif/esp-aws-iot #198: ota_cbor.c: "'cbor_value_get_int' from incompatible pointer type" compilation warnings

krjeschk, 2023-09-13:
I encountered compilation warnings when using _esp-aws-iot ota-for-aws-iot-embedded-sdk_:
[image]

The problems are (as the wanings suggest) incorrect castings here:
[ota_cbor.c#L135](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/f9760892ba152f2c9104d08192ea5ffbbf9fa8ea/source/ota_cbor.c#L135)
[ota_cbor.c#L154](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/f9760892ba152f2c9104d08192ea5ffbbf9fa8ea/source/ota_cbor.c#L154)
[ota_cbor.c#L173](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/f9760892ba152f2c9104d08192ea5ffbbf9fa8ea/source/ota_cbor.c#L173)

They are solved in the newest version of [aws/ota-for-aws-iot-embedded-sdk](https://github.com/aws/ota-for-aws-iot-embedded-sdk/tree/main), but _esp-aws-iot ota-for-aws-iot-embedded-sdk_ uses the old commit of _ota-for-aws-iot-embedded-sdk_

Is is possible to update the _ota-for-aws-iot-embedded-sdk_ submodule in the newest _esp-aws-iot_ release?

n8o, 2025-04-03:
Any updates on this issue?

BastienGermond, 2025-04-03:
This should be fixed with #238 when it's merged.

cmorganBE, 2025-06-28:
Is this fixed now with https://github.com/espressif/esp-aws-iot/commit/f9bccff19a831b9ff073cb2f4431bc2b1a3626d5?