espressif/esp-aws-iot #203: failed to compile the esp-aws-iot-master\examples\mqtt routine

LHYYSY, 2023-11-01:
The following is the log
-- Found Git: D:/idf5.1.1/Espressif/tools/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/idf5.1.1/Espressif/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/idf5.1.1/Espressif/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/idf5.1.1/Espressif/tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
Processing 2 dependencies:
[1/2] espressif/esp_secure_cert_mgr (2.4.0)
[2/2] idf (5.1.1)
CMake Error at D:/idf5.1.1/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    D:/idf5.1.1/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  C:/Users/admin/Desktop/esp-aws-iot-master/libraries/backoffAlgorithm/CMakeLists.txt:2
  (include):

    include could not find requested file:

      C:/Users/admin/Desktop/esp-aws-iot-master/libraries/backoffAlgorithm/backoffAlgorithm/backoffAlgorithmFilePaths.cmake

  Call Stack (most recent call first):

    D:/idf5.1.1/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    D:/idf5.1.1/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

Call Stack (most recent call first):
  D:/idf5.1.1/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/build.cmake:574 (__component_get_requirements)
  D:/idf5.1.1/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:12 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/admin/Desktop/esp-aws-iot-master/examples/mqtt/tls_mutual_auth/build/CMakeFiles/CMakeOutput.log".

gwentech, 2023-11-01:
I saw this too.  For whatever reason others could explain, pulling the repo skips files within some of the library subdirectories.  I manually downloaded each missing portion and dropped it in to my clone, which fixed it.  Guessing there is a more elegant way, but this worked.

LHYYSY, 2023-11-06:
E (342434) coreMQTT: A single byte was not read from the transport: transportStatus=-1.
E (342434) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed.
E (342434) coreMQTT: MQTT connection failed with status = MQTTRecvFailed. 
E (342444) coreMQTT: Connection with MQTT broker failed with status MQTTRecvFailed. 
I am not sure what caused this error

MkSafavi, 2023-11-09:
Did you clone the repository with `--recursive` ?
The libraries are pulled from other git repos (checkout `.gitmodules` file). 
There's no need to download them manually. Performing a recursive pull will download them as well.

LHYYSY, 2023-11-09:
Hi, thank you very much, I've almost solved the problem, or at least been able to run it completely

closed 2023-11-09 by LHYYSY

zdh00718173, 2023-12-12:
> Hi, thank you very much, I've almost solved the problem, or at least been able to run it completely

Then what is your reason that caused the above error you mentioned?

LHYYSY, 2023-12-12:
> > Hi, thank you very much, I've almost solved the problem, or at least been able to run it completely
> 
> Then what is your reason that caused the above error you mentioned?

You have to pull down the code under the subwarehouse that you need

kongshuiJ, 2024-01-24:
> E (342434) coreMQTT: A single byte was not read from the transport: transportStatus=-1.E （342434） coreMQTT：未从传输中读取单个字节：transportStatus=-1。 E (342434) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed.E （342434） coreMQTT：CONNACK recv 失败，状态 = MQTTRecvFailed。 E (342434) coreMQTT: MQTT connection failed with status = MQTTRecvFailed.E （342434） coreMQTT：MQTT 连接失败，状态 = MQTTRecvFailed。 E (342444) coreMQTT: Connection with MQTT broker failed with status MQTTRecvFailed.E （342444） coreMQTT：与 MQTT 代理的连接失败，状态为 MQTTRecvFailed。 I am not sure what caused this error我不确定是什么原因导致了这个错误

Then what is your reason that caused the above error you mentioned?

kongshuiJ, 2024-01-24:
> E (342434) coreMQTT: A single byte was not read from the transport: transportStatus=-1. E (342434) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed. E (342434) coreMQTT: MQTT connection [...]
Hi @LHYYSY 
I have encountered the same problem. How did you solve it?