espressif/esp-aws-iot #211: cbor not present in ESP-IDF >4.3

monkeytronics, 2024-01-18:
I have added esp-aws-iot as a submodule to my project. The modules are being picked up correctly by CMake but the compilation fails with the error message below. 

I am using ESP-IDF 5.1.2 & a recent master branch commit of this library.  The error message regarding cbor led me to hunt around and find that espressif have moved cbor from ESP-IDF to the ESP-IDF component registry. 

[Removed or Deprecated Stuff](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/removed-components.html)

The [advice for cbor](https://components.espressif.com/components/espressif/cbor/versions/0.6.0~1?language=en) is that if you want to use it, you need to install it separately: 

idf.py add-dependency "espressif/cbor^0.6.0~1"

I'm not sure if you have this in your docs anywhere. If not, might be a good idea to add it in. I'm pretty sure I came across some other people posting about cbor issues that may relate to this change.

monkeytronics, 2024-01-18:
On further inspection of this, I can see where cbor is being included in the CMakeLists.txt in ota-for-aws-iot-embedded-sdk. When I added the submodule, I obviously ran :

`$ git submodule --init --recursive`

However, I can see that the tinycbor repo is not there at all! So, it's not surprising that the build was failing. I've manually pasted the repo into place and rerun the build. Next it fails whinging about:

`/libs/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:275:5: error: unknown type name 'mbedtls_threading_mutex_t'`

So I go and have a look around in this repo, and notice that there are a bunch of dependencies missing there too??? Including mdedtls!!! But that kind of makes sense since mbedtls is included as part of ESP_IDF....

Is there some limitation to how this library should be used in order that dependencies like this one work? FWIW, I put all my 3rd party submodules into a folder (outside of any individual project) and pull them in as needed. Shouldn't CMake ensure that the reference to mbedtls is satisfied since it is clearly present in the build???

monkeytronics, 2024-01-18:
Sorry, last post on this. I found another issue which covered the mbedtls issue [link](https://github.com/espressif/esp-aws-iot/issues/168#issuecomment-1897873019). So, with that, we can close these, but I do have several suggestions to save future headaches:
1. Can you add an idf_component.yaml to force the cbor component to get added in by default?
2. Anyone able to explain what the mbedtls fix was? Or how it could be integrated as a default?

nujufas, 2025-03-22:
@monkeytronics , did you manage to fix this?