espressif/esp-aws-iot #212: Connection with MQTT broker failed

kongshuiJ, 2024-01-24:
idf version: **V4.4.2**
esp-aws-iot version: **release/202210.01-LTS**
example: **esp-aws-iot/examples/mqtt/tls_mutual_auth**
platform: **esp32**

I use "xx" instead of "xx" for some sensitive information.

I used  "openssl s_client -connect xxxxxxxxxxxxxx.iot.us-east-2.amazonaws.com:8443 -CAfile root_cert_auth.crt -cert client.crt -key client.key" Verify that the following configurations are correct:
1. root_cert_auth.crt
2. client.crt
3. client.key

AWS Endpoint Hostname and Client ID are also set correctly.
The ssid and password are also set correctly, and the IP address allocation is normal.

But when ESP32 runs, the error result is as follows:

```
I (5097) esp_netif_handlers: example_connect: sta ip: xx.xx.xx.xx, mask: xx.xx.xx.0, gw: xx.xx.xx.xx
I (5097) example_connect: Got IPv4 event: Interface "example_connect: sta" address: xx.xx.xx.xx
I (5107) example_connect: Connected to example_connect: sta
I (5107) example_connect: - IPv4 address: xx.xx.xx.xx
I (5117) coreMQTT: Establishing a TLS session to xxxx-ats.iot.us-east-2.amazonaws.com:8883.
W (5137) wifi:<ba-add>idx:1 (ifx:0, xx:xx:xx:xx:xx:xx), tid:3, ssn:0, winSize:64
E (8397) coreMQTT: A single byte was not read from the transport: transportStatus=-1.
E (8397) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed.
E (8407) coreMQTT: MQTT connection failed with status = MQTTRecvFailed.
E (8407) coreMQTT: Connection with MQTT broker failed with status MQTTRecvFailed.
```

May I ask if anyone has encountered the same problem?

monkeytronics, 2024-01-26:
My issue looks very similar... I'm basically running the subpubwithLED demo from the standard demo project for this library. It seems to be getting stuck at xTlsConnect. I'm looking in it. Will report back here if find solution. 

```
I (15183) monkey_wifi.c: Connected with IP Address:192.168.144.27
I (15183) monkey_wifi.c: Got IP
I (15193) esp_netif_handlers: sta ip: 192.168.144.27, mask: 255.255.255.0, gw: 192.168.144.18
E (15183) XXXX : **** esp_tls_init
E (15203) monkey_wifi.c: monkey task fsm -> MONKEY_WIFI_GOT_IP
E (15213) XXXX : **** esp_tls_conn_new_sync
E (17743) coreMQTT: A single byte was not read from the transport: transportStatus=-1.
E (17743) coreMQTT: CONNACK recv failed with status = MQTTRecvFailed.
E (17753) coreMQTT: MQTT connection failed with status = MQTTRecvFailed.
```

monkeytronics, 2024-01-26:
I've figured out what the issue is in my case. When using the example code, you set

```
/**
 * @brief The thing name of the device.
 */
#define configCLIENT_IDENTIFIER                         ( "S900001" )
```

in CORE_MQTT_AGENT_MANAGER_CONFIG_H. Obviously, this is of zero use to anyone with more than one device, so I changed this to pull the device Unique Serial Number from NVS and loaded this into the Network Context. However, I was caught out because `configCLIENT_IDENTIFIER` is inexplicably referenced in private function (prvCoreMqttAgentConnect) inside core mqtt agent manager.c. I think it was put there as a booby trap just to cost us all a few hours of our life.

kongshuiJ, 2024-02-01:
Hi @monkeytronics 

Thank you very much for sharing. 

My problem has also been resolved, but I only used a new ssid to connect, so I am still not sure why.

closed 2024-02-01 by kongshuiJ

MahadiHasantauhid, 2024-07-05:
> Hi @monkeytronics
> 
> Thank you very much for sharing.
> 
> My problem has also been resolved, but I only used a new ssid to connect, so I am still not sure why.

what have you done exactly? did you only changed the configCLIENT_IDENTIFIER as "SSID" or used new WIFI connection to get rid of the issue?