espressif/esp-aws-iot #213: coreMQTT: Argument cannot be NULL: pContext=0x3ffc3dd8, pSubscriptionList=0x0 AND coreMQTT: MQTT operation failed with status MQTTBadParameter

monkeytronics, 2024-01-27:
Completely Stuck.

Errors = 
```
coreMQTT: Argument cannot be NULL: pContext=0x3ffc3dd8, pSubscriptionList=0x0 
coreMQTT: MQTT operation failed with status MQTTBadParameter
```

Built up demo project. Connecting and publishing to broker fine.  But when I try to subscribe, it breaks with the above error. Full log :

```
I (46777) coreMQTT: MQTT connection established with the broker.
I (46777) core_mqtt_agent_manager: Session present: 0

I (46787) MONKEY_AWS_IOT: coreMQTT-Agent connected.
I (46787) core_mqtt_agent_manager: coreMQTT-Agent connected.
W (46787) MONKEY_AWS_IOT: entering monkey_aws_iot_subscribe - do loop
W (46797) MONKEY_AWS_IOT: monkey_aws_iot_subscribe - do loop - mqttStatus = 0
W (46797) X: entering MQTTAgentCommand_Subscribe
W (46817) X: entering MQTT_Subscribe
E (46817) coreMQTT: * Argument cannot be NULL: pContext=0x3ffc3dd8, pSubscriptionList=0x0.
W (46827) MONKEY_AWS_IOT: prvSubscribeCommand_cb
W (46827) MONKEY_AWS_IOT: prvSubscribeCommand_cb - xTaskToNotify
E (46837) coreMQTT: MQTT operation failed with status MQTTBadParameter

I (46847) MONKEY_AWS_IOT: coreMQTT-Agent disconnected. Preventing coreMQTT-Agent commands from being enqueued.
I (46847) core_mqtt_agent_manager: TLS connection was disconnected.
I (46857) core_mqtt_agent_manager: coreMQTT-Agent disconnected.
W (46867) core_mqtt_agent_manager: starting xTlsConnect
I (49167) coreMQTT: MQTT connection established with the broker.
I (49167) core_mqtt_agent_manager: Session present: 0

I (49167) MONKEY_AWS_IOT: coreMQTT-Agent connected.
I (49177) core_mqtt_agent_manager: coreMQTT-Agent connected.
```

I've done my best to trace the error back. The first error (Argument cannot be empty) comes from core_mqtt::validateSubscribeUnsubscribeParams. And the param in question is pSubscriptionList. 

I've traced this back through

- core_mqtt::MQTT_Subscribe
- core_mqtt_agent_command_functions::MQTTAgentCommand_Subscribe
- processCommand (pulling function from MQTT_AGENT_FUNCTION_TABLE)
- MQTTAgent_CommandLoop
- prvMQTTAgentTask

It's a very long chain. And it appears to me that the offending parameter is created here:

```
MQTTStatus_t MQTTAgent_CommandLoop( MQTTAgentContext_t * pMqttAgentContext )
{
    MQTTAgentCommand_t * pCommand;
    MQTTStatus_t operationStatus = MQTTSuccess;
    bool endLoop = false;

    /* The command queue should have been created before this task gets created. */
    if( ( pMqttAgentContext == NULL ) || ( pMqttAgentContext->agentInterface.pMsgCtx == NULL ) )
    {
        operationStatus = MQTTBadParameter;
    }

    /* Loop until an error or we receive a terminate command. */
    while( operationStatus == MQTTSuccess )
    {
        /* Wait for the next command, if any. */
        pCommand = NULL;
        ( void ) pMqttAgentContext->agentInterface.recv(    <--- what does this do????
            pMqttAgentContext->agentInterface.pMsgCtx,
            &( pCommand ),
            MQTT_AGENT_MAX_EVENT_QUEUE_WAIT_TIME
[... 105 lines cut ...]
         * callback. */
        xSubscriptionAdded = addSubscription( ( SubscriptionElement_t * ) xGlobalMqttAgentContext.pIncomingCallbackContext,
                                              pxSubscribeArgs->pSubscribeInfo->pTopicFilter,
                                              pxSubscribeArgs->pSubscribeInfo->topicFilterLength,
                                              prvIncomingPublish_cb,
                                              ( void * ) ( pxCommandContext->pArgs ) );

        if( xSubscriptionAdded == false )
        {
            ESP_LOGE( TAG,
                      "Failed to register an incoming publish callback for topic %.*s.",
                      pxSubscribeArgs->pSubscribeInfo->topicFilterLength,
                      pxSubscribeArgs->pSubscribeInfo->pTopicFilter );
        }
    }

    if( pxCommandContext->xTaskToNotify != NULL )
    {
        ESP_LOGW(TAG, "prvSubscribeCommand_cb - xTaskToNotify");
        xTaskNotify( pxCommandContext->xTaskToNotify,
                     ( uint32_t ) ( pxReturnInfo->returnCode ),
                     eSetValueWithOverwrite );
    }
}
```

monkeytronics, 2024-01-29:
Closing as this is related to demo code, not the actual library itself. Root issue in this case was task priorities in demo app tasks conflicting with rest of user application.

closed 2024-01-29 by monkeytronics

willcode4food, 2024-12-30:
@monkeytronics  I'm stuck also in the same spot and am scratching my head what happens to `pCommand` in the `recv` function.   I believe this ultimately is a reference to [Agent_MessageReceive](https://github.com/espressif/esp-aws-iot/blob/cd1bd3b654f7cf6393bb92100bb7916cf9152c06/libraries/coreMQTT-Agent/port/freertos_agent_message.c#L49).  But my issue is that the `pTopicFilter` variable that is empty and I get an error "Subscription #0 in SUBSCRIBE packet cannot be empty."  I can't seem to find where I loose the reference to the topic filter.  But I narrowed it down to `MQTTAgent_CommandLoop`.   Any help would be much appreciated.