espressif/esp-aws-iot #216: Fleet Provisioning build error

amitabhskrishworks, 2024-04-02:
Hello there,
I have cloned the esp-aws-iot folder using the "https://github.com/espressif/esp-aws-iot.git" .Without making any changes I tried to build the fleet_provisioning_with_csr example (given at: esp-aws-iot/examples/fleet_provisioning/fleet_provisioning_with_csr) but got stuck with build errors (mentioned below) but was able to build the mqtt/tls_mutual_auth example. Although both the examples are in same folders but i don't know why is fleet one not working.

C:/Users/Amitabh/esp/esp-aws-iot/libraries/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/source/fleet_provisioning.c:765:1: fatal error: opening dependency file esp-idf\Fleet-Provisioning-for-AWS-IoT-embedded-sdk\CMakeFiles\__idf_Fleet-Provisioning-for-AWS-IoT-embedded-sdk.dir\f7893e654caf0f5cdf916ac2a3b97995\fleet_provisioning.c.obj.d: No such file or directory
  765 | }
      | ^
compilation terminated.
[937/947] Building C object esp-idf/corePKCS11/CMakeFiles/__idf_corePKCS11.dir/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. 

Detailed log are in the file
[fleet_prov_log_2Apr2024.txt](https://github.com/espressif/esp-aws-iot/files/14833047/fleet_prov_log_2Apr2024.txt)

ookamipjq, 2024-04-09:
I also encountered the same problem, have you solved it?

PaulAnurag, 2024-05-28:
Or else you can just change the name of "Fleet-Provisioning-for-AWS-IoT-embedded-sdk" to "Fleet-Provisioning".

This is easy and best solution if enabling long path does not work

amitabhskrishworks, 2024-05-28:
If change the branch to any of these then it works
[release/202012.04-LTS](https://github.com/espressif/esp-aws-iot/tree/release/202012.04-LTS)
[release/202210.01-LTS](https://github.com/espressif/esp-aws-iot/tree/release/202210.01-LTS)

amitabhskrishworks, 2024-05-28:
> Hi @amitabhskrishworks Issue is due to maximum filename limitation in Windows filesystem.
> 
> Can you try enabling longpaths on your Windows as instructed [here](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=powershell#tabpanel_1_powershell).
> 
> If this doesn't work as well, can you try placing esp-aws-iot somewhere in C:\ directly for now? We will see how we can shorten the filepath of the object file for Windows. Unfortunately. we don't have a way around it as of yet.

This solution worked for me 
Thank you

closed 2024-05-28 by amitabhskrishworks