espressif/esp-aws-iot #217: Any esp-aws-iot release supporting or usable with IDF v5.2.1 ?

espressing, 2024-04-11:
Hi,

as IDF v5.2.1 has been stable release with many needed fixes & changes, we would take it into use.

However, it seems esp-aws-iot releases (not even master branch) support it? And seen some ticket about this that only solved it on master and only for IDF5.1.

Is there any way of using a stable esp-aws-iot version with IDF5.2.1?  or we need to go back to IDF5.1 and use esp-aws-iot from master branch that then may have issues?

Thanks in advance.

espressing, 2024-04-25:
Hi @avsheth 
Many thanks, this is very good news - thanks in advance for a timeline;  
Kind regards

espressing, 2024-05-08:
Hi @avsheth
Could you please inform whether there are any updates on mentioned timeline?

espressing, 2024-05-08:
> Hi @espressing <https://github.com/espressing>
> Yes we will announce the support by 17th.

Hi,
excellent, many thanks!

Jona2892, 2024-05-17:
Hey,
Was the support announced today? If not, when will it be ready?

Best

espressing, 2024-05-17:
Hi @avsheth,

thank you, very good news! So presumably master branch only or the above means even the latest stable tagged release is OK?

espressing, 2024-05-17:
> We plan to support IDF v5.2.1 only on the master branch for now.

Thank you!