espressif/esp-aws-iot #219: MQTT_ProcessLoop() producing 1 s Pings even with 40s KeepAlive time.

SCgenie, 2024-05-14:
Hi,

MQTT_ProcessLoop() producing 1s Pings every time it is called. Anyone know how to avoid it. I am providing correct milli second time callback. 

A workaround is using MQTT_ReceiveLoop() and self managed MQTT_Ping() at keepalive interval. This works great, however, these two functions return SUCCESS even when I pull the WAN cable. So I cannot detect a MQTT broker/server disconnect.

Any suggestions?

Thanks

SolidStateLEDLighting, 2024-05-17:
Based on my experience, it takes some time (a number of seconds) for the system to detect the loss in Internet access and return a failure code.

Your information about ReceiveLoop and Ping calls is interesting.     I have only used the ProcessLoop call at this point.

SCgenie, 2024-05-17:
Problem was in core_mqtt.c , line 1356
            const uint32_t timeElapsed = calculateElapsedTime( now, pContext->lastPacketRxTime );

            if( ( timeElapsed != 0U ) && ( timeElapsed >= PACKET_RX_TIMEOUT_MS ) )
            {
                status = MQTT_Ping( pContext );
            }

pContext->lastPacketRxTime; this is not updated ever and causes ping at whatever rate you are calling MQTT_ProcessLoop(). I see the newer versions of core_mqtt.c handles it in the main branch. No new releases with that yet.

Another thing to be aware of is that the defaults for these limit your KeepAlive time to 30s,
PACKET_TX_TIMEOUT_MS 30s
PACKET_RX_TIMEOUT_MS 30s
in  #include "core_mqtt_config_defaults.h"

SCgenie, 2024-05-17:
> Based on my experience, it takes some time (a number of seconds) for the system to detect the loss in Internet access and return a failure code.

Server drops you typically 1.5x KeepAlive sent to it during MQTT_Init(). I can see MQTT_ProcessLoop() return error in 4 calls at 700mS interval.

SolidStateLEDLighting, 2024-05-17:
Thanks for the info.  Your investigation has value.