espressif/esp-aws-iot #230: Memory allocation problem occurs in AWS IoT OTA process (IDFGH-13815)

Gyunamchoi, 2024-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3 2nd stage bootloader

### Espressif SoC revision.

ESP32-WROOM-32UE chip revision: v3.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board (ESP32-WROOM-32UE)

### Power Supply used.

USB

### What is the expected behavior?

The ultimate goal is to proceed with OTA updates through aws-iot.

### What is the actual behavior?

An error related to heap_caps occurs when allocating memory such as malloc.

### Steps to reproduce.

1. Power on 
2. AWS IoT connection and MQTT connection via SIM7600EK modem with lwip(ppp) interface
3. Transmitting shadow and measurement information - works well
4. Create ota job in aws iot
5. AWS -> device : topic : $aws/things/(device id)/jobs/notify-next (OK)
6. device -> AWS : topic : $aws/things/(device id)/jobs/start-next (OK)
7.  AWS -> device : topic : $aws/things/testdev21/jobs/start-next/accepted (OK)
8. device -> AWS : $aws/things/(device id)/streams/(stream_id)/get/json ( message Publish ERROR ) 

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d1c6  PS      : 0x00060b33  A0      : 0x8008cb64  A1      : 0x3ffc40e0  
0x4008d1c6: remove_free_block at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:332
 (inlined by) block_locate_free at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:1015

A2      : 0x3ffd039c  A3      : 0x00000058  A4      : 0x00000002  A5      : 0x00000036  
A6      : 0x3ffc4c1c  A7      : 0x3ffb55f4  A8      : 0x00000006  A9      : 0xffffffff  
A10     : 0x00000003  A11     : 0xffffffff  A12     : 0x00000468  A13     : 0x3ffb5618  
A14     : 0x3ffc4240  A15     : 0x0000cdcd  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000b  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
0x4000c349: memcpy in ROM
[... 135 lines cut ...]
//                        char* msg_mqtt_start = ota_request_start_stream(json_otadocument);
                        ESP_LOGI(TAG, "topic_mqtt_start =%s  msg_mqtt_start = %s ", topic_mqtt_start, msg_mqtt_start);                    
                        **mqtt_publish(topic_mqtt_start, msg_mqtt_start);****
                        ESP_LOGI(TAG, "Start ota [%s] from factory mode to stream image file request complete.",stream_id);
                        free(topic_mqtt_start);
                        free(msg_mqtt_start);
                    } else {
`
mqtt message process
1. When an MQTT event occurs, the event handler processes the event and passes the topic and data as char* arguments.
2. Parse the topic
3. If it is jobs, pass it to the jobs process
4. Receive the $aws/things/(device id)/jobs/notify-next topic and check if the status is queued.
5. Once the check is complete, publish the $aws/things/(device id)/jobs/start-next message
7. Then, AWS checks that the message status of the $aws/things/testdev21/jobs/start-next/accepted topic is IN PROGRESS as the device. 8. Once confirmed, publish the $aws/things/(device id)/streams/(stream_id)/get/json message to AWS
(In this process, a kernel panic occurs during memory allocation)

When creating cJSON, cJSON_CreateObject() uses malloc, so a kernel panic occurs, and the same problem occurs in esp_mqtt_publish() when publishing. A heap_caps_calloc error occurs when performing memory-related operations.
Whether creating with malloc, creating with fixed generation, or using cJSON, the same problem occurs in the same part.

There is about 170KB of free heap memory, and there are no problems at all in periodic message publishing, except for the jobs-related part.

I have no idea whether it is a problem with the mqtt module itself when publishing, a problem with another part, or a problem with json. Please help me.

Thank you.

kvp1703, 2024-11-11:
Hi @Gyunamchoi,
Can you help share the esp-aws-iot branch you're on?
Also, can you try with standard [OTA example](https://github.com/espressif/esp-aws-iot/tree/202210.01-LTS-release/examples/ota/ota_mqtt) ?
We looked at your code, could you verify whether recv_topic and recv_data are being properly freed, including edge cases? If not, please provide the relevant code snippet that handles the freeing or utilization of these resources.

Gyunamchoi, 2024-11-11:
@kvp1703 Oh I'm Sorry.
It's been resolved, but I didn't close it.
There was a subtle mismatch in the size and timing when allocating, using, and freeing memory.
It's resolved now and it works fine.
Thank you for your interest.

closed 2024-11-11 by Gyunamchoi