espressif/esp-aws-iot #235: Any esp-aws-iot release (or master branch) usable with IDF v5.4?

espressing, 2025-01-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

further to what is listed as compatible IDF release with the (at least master branch) of the esp-aws-iot codeline:

Currently only IDF5.2 is listed as compatible - could you please confirm whether _the recent IDF5.4 (or at least IDF5.3) is compatible / usable with esp-aws-iot  master branch or any release?

As the IDF5.3 and IDF5.4. releases contain many updates/fixes, ideally we'd like to use latest (even if just master branch) esp-aws-iot with latest IDF.

Thanks in advance.

imdahisaria, 2025-03-12:
Any update on this topic ?

scholayil, 2025-03-13:
I am using latest release with IDF v5.4; coreMQTT, coreJSON and a couple of other libraries actvely used.

imdahisaria, 2025-03-13:
> I am using latest release with IDF v5.4; coreMQTT, coreJSON and a couple of other libraries actvely used.

Can you help ? I want to use the fleet provisioning with car example in my project . But I am not able to compile . I cloned the repo in components and tried to link the files in CMakelist.txt but getting lot of missing library errors.  Can you share your project setup and CMakelist contents ? Would be highly appreciated and helpful

scholayil, 2025-03-13:
> Can you help ? I want to use the fleet provisioning with car example in my project . But I am not able to compile . I cloned the repo in components and tried to link the files in CMakelist.txt but [...]

[seethis.txt.txt](https://github.com/user-attachments/files/19237814/seethis.txt.txt)

imdahisaria, 2025-03-14:
> [seethis.txt.txt](https://github.com/user-attachments/files/19237814/seethis.txt.txt)

Thank you for this. I followed your above steps. After that 

1. Compiled -> OK
2. Now in my main folder "com-module.c" under function app_main() , I am calling "aws_iot_demo_main(0,NULL);"
4. Compiling gives the following error

`main/com_module.c:384:(.text.app_main+0x395): undefined reference to `aws_iot_demo_main'`

My project structure is as follows

[image]

I added your changes in the root CMakelist.txt which is selected in the image.

Can you suggest and help how can I resolve this.? Help would be appreciated .

Thank you